libopencm3/libopencm3 #392: stm32 timer_clear_flag() race condition

cdhooper, 2015-01-05:
The following code in lib/stm32/common/timer_common_all.c:

```
void timer_clear_flag(uint32_t timer_peripheral, uint32_t flag)
{
        TIM_SR(timer_peripheral) &= ~flag;
}
```

introduces a race condition when clearing interrupt status flags of a timer.
If another interrupt flag asserts in between the read / modify / write, then
it will be lost.  This is because bits in this register are defined by the (STM32F407)
datasheet as rc_w0, meaning software can write a 1 with no effect, or write a
0 to clear a particular status.  The recommended fix is simple:

```
void timer_clear_flag(uint32_t timer_peripheral, uint32_t flag)
{
        TIM_SR(timer_peripheral) = ~flag;
}
```

There is no need to do the read / modify.  All bits in the register are rc_w0,
aside from reserved bits.  It is presumable that any future bits ST assigns
to this register would also be rc_w0.

karlp, 2015-04-29:
I agree, that appears to be a needless RMW.  Nice catch.  However, reserved bits are "must be kept at reset value" So, in the strictest sense, you must do the RMW.  I'm curious if anyone has an insider knowledge from ST on what you can _really_ do for reserved bits, but I'd be inclined to agree with you that that the straight write is probably perfectly safe.

cdhooper, 2015-04-29:
Understood -- it's better to follow the datasheet to the letter if we are striving for correctness.  Another implementation which avoids the interrupt status clear race condition and preserves the reserved bits:

void timer_clear_flag(uint32_t timer_peripheral, uint32_t flag)
{
        TIM_SR(timer_peripheral) = ~flag | (TIM_SR(timer_peripheral) & 0xffffe0c0);
}

karlp, 2015-04-29:
a) that mask is only going to be appropriate for specific timers, and b) how is that any safer?

zyp, 2015-04-29:
The «must be kept at reset value» seems like sort of a generic disclaimer because they are too lazy to distinguish between secret special purpose bits and bits that are simply ignored. In practice, pretty much all reserved bits are of the latter sort. For a status-register containing only r and rc_w0 bits, it should be safe to assume that you can write 1 to everything you don't want to clear.

cdhooper, 2015-04-29:
I agree zyp.  In fact, I just looked at ST-Micro's own reference driver and they do exactly as I suggested above -- SR = ~flag.  The only difference is that their access is 16-bit.  If ST's driver does it, then they can't change the architecture without breaking their own code.

karlp, I am looking at the generic timers for STM32F405/F407,STM32F415/Ff14, STM32F42 reference manual in 18.4.5 and also Table 99.  You are right that other timers will have more reserved bits, so it doesn't eliminate the concern for those bits.

What my original suggestion does eliminate is the race condition where interrupts can be missed.  The race is created because |= is not an atomic operation.  When compiled, the assembly for the function is as follows:
   0x08050ee6 <+0>:	ldr	r3, [r0, #16]
   0x08050ee8 <+2>:	bic.w	r1, r3, r1
   0x08050eec <+6>:	str	r1, [r0, #16]
   0x08050eee <+8>:	bx	lr

Load, bit clear, store.  If a lower priority interrupt occurs between the bit clear and store, its status will be lost because of the rc_w0 property of the register.  Losing an interrupt, in my opinion, is way worse than the concern over preserving reserved bits.

karlp, 2015-04-29:
Yes, I understand how your original solution would fix the race, I was curious the second version possibly could :)

If you can make a pull request for this, that would be great.  It should cover all the timer interrupts at least.

fenugrec, 2015-07-27:
I can make a PR for the timer issue. I did a very quick check on other stm32 code and noticed a similar issue with stm32/f3/adc.c  (R-M-W on ADC_ISR)

karlp, 2016-03-30:
closed in pull #492

closed 2016-03-30 by karlp

neoxic, 2022-01-11:
> Load, bit clear, store. If a lower priority interrupt occurs between the bit clear and store, its status will be lost because of the rc_w0 property of the register. Losing an interrupt, in my opinion, is way worse than the concern over preserving reserved bits.

Please correct me if I am wrong, but I don't see how a "lower priority interrupt" can occur and be lost. We are talking about **one** interrupt flag. So it's rather **the same** interrupt can occur and be lost (due to an unrelated high priority interrupt happening inside an interrupt handler) because both `TIM1_SR &= ~BIT` and `TIM1_SR = ~BIT` statements clear only **one** bit albeit the former makes an unnecessary load whereas the latter does not.

In other words, please consider the following code where we need to determine which interrupt we are serving:
```C
void tim1_brk_up_trg_com_isr(void) {
    if (TIM1_SR & TIM_SR_UIF) { // Update
        // <-------- if another update event is fired here, it is lost
        TIM1_SR = ~TIM_SR_UIF;
    }
    if (TIM1_SR & TIM_SR_TIF) { // Trigger
        // <-------- if another trigger event is fired here, it is lost
        TIM1_SR = ~TIM_SR_TIF;
    }
}
```
Despite both interrupt flags are cleared atomically, we can still lose pending interrupts because there's a race between checking for a flag and clearing it. Any ideas?

cdhooper, 2022-01-11:
This thread has been closed over five years, so please pardon my poor memory. What I think I meant by "lower priority interrupt" in the follow-up comment was simply another interrupt flag could be set. 

Yes, there is only one interrupt, but there are potentially 13 different interrupt flags which can trigger that one interrupt. The original bug report was against timer_clear_flag(), which can clear one or more flags.

Please ignore the possibility of a higher priority interrupt occurring at any point in time. I think this is clouding your understanding of the actual problem of this bug report. The issue was that the register is Write-Zero-to-Clear.

Looking at your specific example:
    TIM1_SR = ~TIM_SR_UIF;
    
This is correct code. It is writing a Zero value to _only_ the timer status flag which needs to be cleared. (~TIM_SR_UIF = 0xfffffffe). If the code were instead:
    TIM1_SR &= ~TIM_SR_UIF;
    
Then this is incorrect code because in between the LDR of TIM1_SR, the BIC, and the STR, another status flag could get asynchronously set by hardware. Doesn't matter if a different higher priority interrupt occurs at that point or not, the loaded value of TIM1_SR is now out of date relative to what is being written. Imagine this timeline:
    Hardware sets UIF in TIM1_SR.
    The NVIC generates an interrupt in the processor which then executes our interrupt service routine.
    After capturing the current value of TIM1_SR, the function calls the broken timer_clear_flag() to clear TIM_SR_UIF.
    timer_clear_flag()   (the old broken version) executes:
        LDR
                   Hardware sets CC1IF in TIM1_SR
        BIC
        STR

In the above, no secondary interrupt was needed to inappropriately clear CC1F, yet it was cleared because the value for that bit, as captured by the LDR, was zero. When the STR happened, that bit was cleared, losing the CC1IF flag.

I think your specific concern regarding higher priority interrupts occurring during an interrupt handler is actually a completely separate issue from what was reported in this bug.

Looking at the provided example function:
```
void tim1_brk_up_trg_com_isr(void) {
    if (TIM1_SR & TIM_SR_UIF) { // Update
        // <-------- if another update event is fired here, it is lost
        TIM1_SR = ~TIM_SR_UIF;
    }
    if (TIM1_SR & TIM_SR_TIF) { // Trigger
        // <-------- if another trigger event is fired here, it is lost
        TIM1_SR = ~TIM_SR_TIF;
    }
}
```

The problem, as I think you are describing it, is that a second event of the same type could occur between checking for the flag and clearing it. Of course it certainly could. That second event of the same type *could* also occur even before the code even checks for the flag. It's lost, and there's no going back on that if your handler could not process the flag in time before a second flag is asserted. But what you do have in either case is that your code is executing the handler for that flag. It can fully process the UIF or TIF knowing that at least one of them occurred.

Now perhaps you have code that looks like this (simplifying slightly):

``` 
void tim1_brk_up_trg_com_isr(void) {
    if (TIM1_SR & TIM_SR_UIF) { // Update
        // Code to service UIF is here
        // <-------- if another update event is fired here, it is lost
        TIM1_SR = ~TIM_SR_UIF;
    }
}
```

This code is potentially broken because during the time of servicing the UIF, another update event could occur, and then the UIF flag that there is more work to do has been lost. Maybe this is what you meant. To avoid this, I always clear flags I will service at the top of a handler, or that flag's handler. Example:

``` 
void tim1_brk_up_trg_com_isr(void) {
    uint32_t status = TIM1_SR;
    if (status & TIM_SR_UIF) { // Update
        TIM1_SR = ~TIM_SR_UIF;
        // Code to service UIF is here
    }
}
```

Then if another update even happens while servicing the previous UIF, the TIM_SR_UIF flag will be set again and another interrupt triggered when this interrupt handler returns. I hope this helps. If you need further help or clarification, please message me directly rather than commenting on this old bug report.

neoxic, 2022-01-12:
@cdhooper Thank you for the detailed input! I can see the original problem now. The part that I missed in the broken version of `timer_clear_flag` was indeed that an old value was captured with more zeroes than needed. Thanks again!

As to your last code example, this is exactly the way I handle interrupts. I save the status and clean it bit by bit. And I agree with you on that one - since we are already servicing a certain flag, there's no way to distinguish whether it has been raised again or not while in the handler. Except when there's hardware support for that overcapture being an example.

neoxic, 2022-01-12:
Interestingly though, saving (caching) the status register before checking it against set flag bits, as in your last example, is not the best way to service interrupts as fast as possible. Because potentially, an interrupt bit that the routine is capable of servicing may get set in the middle of it. Hence reading the status every time before checking (like in my `tim1_brk_up_trg_com_isr` example) could potentially save you another interrupt routine entry.🙂

cdhooper, 2022-01-12:
You may be right, but that depends on the use case and how frequently you expect multiple interrupts to near-simultaneously occur, and in what order they occur. An STM32 peripheral device read/write is always slower than a register access, so repeatedly banging a peripheral register has consequences.

Take this handler code for example:
```
if (TIM1_SR & TIM_SR_UIF) {
    TIM1_SR ~TIM_SR_UIF;
    // process
}
if (TIM1_SR & TIM_SR_CC1IF) {
    TIM1_SR ~TIM_SR_CC1IF;
    // process
}
if (TIM1_SR & TIM_SR_CC2IF) {
    TIM1_SR ~TIM_SR_CC2IF;
    // process
}
if (TIM1_SR & TIM_SR_CC3IF) {
    TIM1_SR ~TIM_SR_CC3IF;
    // process
}
if (TIM1_SR & TIM_SR_COMIF) {
    TIM1_SR ~TIM_SR_COMIF;
    // process
}
```

On every interrupt, this code is doing at least five reads of TIM1_SR and at least one write. All for the hope of catching another status flag during the (hopefully short) time spent in interrupt context. It might catch it, and it might be a win, but in general, this code is consuming more CPU time than it should in interrupt context. It's also somewhat weakened by the fact that it's not even catching all possible cases. For example, if a UIF occurs while processing CC1IF, the code is not going to go back and handle UIF without adding a loop or re-entering the interrupt handler.

If you really worry about the re-entry time of the interrupt handler, you can still have the benefit of your idea, not wasting so many processor cycles, and not missing (most) late occurring secondary events. Example:

```
uint32_t status;

while ((status = TIM1_SR) != 0) {
    TIM1_SR = ~status;
    if (status & TIM_SR_UIF) {
        // process
    }
    if (status & TIM_SR_CC1IF) {
        // process
    }
    if (status & TIM_SR_CC2IF) {
        // process
    }
    if (status & TIM_SR_CC3IF) {
        // process
    }
    if (status & TIM_SR_COMIF) {
        // process
    }
}
```
There is still waste happening here (minimum two reads of TIM1_SR and one write). There is always the possibility of still missing late occurring events. At some point the handler needs to return from the ISR. The latency between the final register check and the return from the interrupt exception.

zyp, 2022-01-12:
> Interestingly though, saving (caching) the status register before checking it against set flag bits, as in your last example, is not the best way to service interrupts as fast as possible. Because [...]

Cortex-M does tail-chaining of interrupts, meaning that if there's still active interrupts when an ISR returns, it'll go right on to call the next ISR without a full state unstacking and stacking in between. The overhead of a tail-chained ISR re-entry might very well be lower than polling a status register multiple times, especially considering the re-entry will only happen if there's still active interrupt bits.

neoxic, 2022-01-12:
You are right, gentlemen! And as I said earlier, personally, I prefer to read a status register only once in the beginning of a handler clearing processed flags bit by bit atomically. This approach is the middle ground and looks reasonable for most cases. My note was meant to be just a brief thought experiment.