libopencm3/libopencm3 #409: USB MSC: Writes aren't acknowledged

Spritetm, 2015-02-21:
The USB MSC code has a bug: if a write request is sent, the MSC code will handle it but won't send an CSW back, leaving the host to wait forever. This is actually visible in the examples in the example repo: if you flash a msc example and try to write to the resulting drive, at least under Linux the device hangs.

Here's the fix I used to solve it. I'm not a USB expert by far, but as far as I can tell this fixes the issue:
--- usb_msc.c	2015-01-28 17:38:50.275440725 +0100
+++ usb_msc_mine.c	2015-02-21 23:04:41.681417393 +0100
@@ -586,6 +588,19 @@ static void msc_data_rx_cb(usbd_device *
 				trans->current_block++;
 			}
 		}
+
+		if (false == trans->csw_valid) {
+			scsi_command(ms, trans, EVENT_NEED_STATUS);
+			trans->csw_valid = true;
+		}
+		left = sizeof(struct usb_msc_csw) - trans->csw_sent;
+		if (0 < left) {
+			max_len = MIN(ms->ep_out_size, left);
+			p = &trans->csw.buf[trans->csw_sent];
+			len = usbd_ep_write_packet(usbd_dev, ms->ep_in, p, max_len);
+			trans->csw_sent += len;
+		}
+
 	} else if (trans->byte_count < trans->bytes_to_write) {
 		if (0 < trans->block_count) {
 			if ((0 == trans->byte_count) && (NULL != ms->lock)) {

klenSA, 2015-11-09:
I confirm the problem under Linux, and the decision of the bug above
code
thanks Spritetm

closed 2018-03-02 by karlp

Spritetm, 2018-03-03:
Whoah, that took a while. Thanks for taking care of this!