libopencm3/libopencm3 #431: Any plan for supporting LPC810?

dword1511, 2015-05-09:
LPC810 is a Cortex-M0+ microcontroller in DIP-8 package from NXP. NXP provides some code in this package: http://www.lpcware.com/system/files/lpcopen_2_15_lpcxpresso_nxp_lpcxpresso_812.zip and there is example codes from https://github.com/microbuilder/LPC810_CodeBase

I have a few chips and can help testing.

(I did a search in the issues and found nothing related to the LPC810, sorry if I am wrong)

karlp, 2015-05-09:
It would be gratefully accepted if someone contributed it, though I'm not personally aware of anyone working on it right now.  It shouldn't require much to reuse a lot of the lpc17x and lpc18x code, as I imagine the peripherals are probably similar/same.

karlp, 2015-07-29:
I don't see any great need to keep this open as an "issue" of any sort, but feel very free to send pulls for support of the lpc8xx

closed 2015-07-29 by karlp