libopencm3/libopencm3 #435: Interrupt driven USB for STM32?

dword1511, 2015-05-13:
I saw that libopencm3 has support for interrupt driven USB on LM4F (`usb_enable_interrupts` and `usb_disable_interrupts`).

By enabling and handling the interrupts and removing the polling in the loop, USB seems to be working well:

```
void main(void) {
  ......
  nvic_enable_irq(NVIC_USB_LP_CAN_RX0_IRQ);
  nvic_enable_irq(NVIC_USB_WAKEUP_IRQ);
  ......
}

void usb_wakeup_isr(void) {
  usbd_poll(usbd_dev);
}

void usb_lp_can_rx0_isr(void) {
  usbd_poll(usbd_dev);
}
```

rkreis, 2015-05-14:
Another example without interrupts, but `wfi` (which allows using sleep modes) combined with `usb_poll`: https://github.com/rkreis/usb_testcase
Relevant: #360 and https://github.com/kuldeepdhaka/libopencm3/commit/5f873fc45b5c8c6b8e6fd33557f99d41aa9a9108

karlp, 2015-05-14:
a few people have written it like that, there was some discussion that this was only suitable really in certain test cases, but sure, you can call poll however you like, that's entirely up to you.

schodet, 2015-09-19:
If you do this with OTG IP, see #531.

closed 2019-01-08 by dword1511