libopencm3/libopencm3 #442: MCO on STM32F1

dword1511, 2015-05-20:
I recently tried to enable MCO on a `STM32F103RBT6` device (revision code **Y**). It turns out that

```
  rcc_clock_setup_in_hse_8mhz_out_72mhz();
  rcc_periph_clock_enable(RCC_GPIOA | RCC_AFIO);

  gpio_set_mode(GPIOA, GPIO_MODE_OUTPUT_50_MHZ, GPIO_CNF_OUTPUT_ALTFN_PUSHPULL, GPIO8);
  rcc_set_mco(RCC_CFGR_MCO_HSECLK);
```

works, but the following

```
  rcc_clock_setup_in_hse_8mhz_out_72mhz();
  rcc_periph_clock_enable(RCC_GPIOA | RCC_GPIOB | RCC_AFIO);

  gpio_set_mode(GPIOA, GPIO_MODE_OUTPUT_50_MHZ, GPIO_CNF_OUTPUT_ALTFN_PUSHPULL, GPIO8);
  rcc_set_mco(RCC_CFGR_MCO_HSECLK);
```

does NOT.

The difference is that the latter code **enables clock for GPIOA and GPIOB simultaneously**, while the former one that works **only enables clock for GPIOA**. If afterwards `rcc_periph_clock_enable(RCC_GPIOA)` is called separately, the code also works.

I wonder if this is a bug in the `libopencm3` or it is actually a hardware bug.

Related code can be found [here](https://github.com/dword1511/stm32-freqmeter).

karlp, 2015-05-20:
this is an api documentation problem.  the rcc_periph_clock_enable(x) isn't actually designed to take an 'or'd list of params.  the RCC_GPIOA define is a special magic define that includes both which register, and which bit.  (RCC_AHBENR vs APB1ENR vs APB2ENR and so on)

the older rcc_peripheral_enable_clock(&RCC_AHB1ENR, RCC_AHB1ENR_IOPBEN) form allows 'or'ing the second parameters together _as long as all bits are for the same port_

See http://libopencm3.github.io/docs/latest/stm32f1/html/group__rcc__defines.html#gaaf3dd53c1ced02082fce0076976547a8 
and 
http://libopencm3.github.io/docs/latest/stm32f1/html/group__rcc__defines.html#ga90aa2b7801b2b42debc0536d38c5b07c

See also all the examples in the libopencm-examples repo, for instance: https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f1/stm32vl-discovery/adc-dac-printf/adc-dac-printf.c#L42-L48

If you feel the docs should explicitly state that you can't simply OR parameters together, please re-open this bug.

closed 2015-05-20 by karlp

dword1511, 2015-05-20:
Then is there a function that actually allows ORing? I feel that for some space-constrained devices this could same some space if quite a lot of peripherals need to be enabled. This can be a corner case though.

Although it is better to mention that explicitly in the documentation, I do not think it will help much. I guess I just carried old habits from ST's standard peripheral library over.

Thanks for answering and for the awesome project!

karlp, 2015-05-20:
as I mentioned, "the older rcc_peripheral_enable_clock(&RCC_AHB1ENR, RCC_AHB1ENR_IOPBEN) form allows 'or'ing the second parameters together as long as all bits are for the same port"

If you care about space that much, it's much simpler to just use the registers yourself

ChuckM, 2015-05-20:
@dword1511 you have two choices if you want to OR in various enabled flags. You can store them into 
the register directly, which is to say something like:

`RCC_AHB1ENR |= RCC_AHB1ENR_IOPAEN | RCC_AHB1ENR_IOPBEN | RCC_AHB1ENR_AFIOENR;`

But the convenience api works for any peripheral on any peripheral bus and so it is constrained to do so individually.