libopencm3/libopencm3 #443: Issues with building for STM32

andrey-gvrd, 2015-05-20:
It seems like I'm missing a file when building with a makefile located in lib/stm32/f0:

    [andrey@localhost f0]$ make
    CC      vector.c
    In file included from ../../../include/libopencm3/cm3/nvic.h:135:0,
                 from ../../../include/libopencm3/cm3/vector.h:41,
                 from ../../cm3/vector.c:21:
    ../../../include/libopencm3/dispatch/nvic.h:6:39: fatal error: libopencm3/stm32/f0/nvic.h: No such file or directory
     # include <libopencm3/stm32/f0/nvic.h>
                                       ^
    compilation terminated.
    ./../Makefile.include:44: recipe for target 'vector.o' failed
    make: *** [vector.o] Error 1

Am I doing something wrong?

karlp, 2015-05-20:
make in the root directory, don't go poking around lower down by yourself.  there's headers generated from machine definitions.

closed 2015-05-20 by karlp

amirhammad, 2015-05-20:
For compilation of the specific part of the library, I am using, for example, following command: 
> make lib/stm32/f0

ChuckM, 2015-05-20:
Yes @amirhammad that is incorrect. From the root directory just type 'make' and while it will make all of the various versions it will also complete the very important step of generating specific headers.

karlp, 2015-05-21:
@amirhammad if you found any documentation suggesting that command invocation, please let us know so we can remove it.

amirhammad, 2015-05-21:
@ChuckM
While I am looking at the Makefile(in the root) , Target lib/stm32/f0 is part of the _$(LIB_DIRS): $(IRQ_DEFN_FILES:=.genhdr)_ rule. And hence, I think generation of specific headers does take place there. 
@karlp
It is pure improvisation without any documentation. :P

asafsirin, 2020-12-29:
BTW, the **python3** name for python executable pointed under shebang in header generating scripts isn't used in newer installations of windows python. so either it has to be changed to **python** or **doskey python3=python** must be executed under **msys**. 
looks like msys doesn't properly handle alias command.

karlp, 2020-12-29:
@asafsirin please, there's absolutetly nothing useful in commenting on a closed 5 year old ticket.