libopencm3/libopencm3 #447: USB-F103 Lock up due to control race

kuldeepdhaka, 2015-05-23:
if DIR is set
	if CTR_TX and CTR_RX (both) is set:
		switch(endpoint-type)
		case control:
			if transfer is control-out:
				process CTR_RX
				process CTR_TX
			else transfer is control-in
				process CTR_TX
				process CTR_RX
		default: (other than control)
			process {CTR_TX, CTR_RX} how ever we want, they are two independent endpoints
	else [CTR_RX is only set]:
		process CTR_RX
else [CTR_TX is only set]
	process CTR_TX

note:
endpoint0 is always control
[ ] -> denotes that it is known, no need to test

kuldeepdhaka, 2015-08-19:
@gmccollister statement:
CTR_RX and CTR_TX both become set but the OUT handler is called, resulting
in an endless stall loop. This situation doesn't happen frequently, I haven't
been able to figure exactly why it happens though I'm able to reproduce it
within a few minutes by blasting the device with non-stop control and bulk
endpoint traffic that also generates USART traffic. One theory I have is that
another interrupt is sometimes occuring after LAST_DATA_IN, giving the
STATUS_OUT complete enough time to come in.

test:
since i do not have f1, i tested the code on f0 (everything same, only the PMA is bit different).
https://github.com/kuldeepdhaka/libopencm3-examples/tree/usb/f103-lock-up-due-to-control-race
this is a usb device firmware and python code,
try to send control traffic and bulk traffic in parallel.

results:
when the uC is running at 8Mhz, the uC is not able to keep up with the incoming traffic and in both case (same with patch and without patch) there is equal error rate.

when the uC is running at 16Mhz, 24Mhz, 48Mhz, there is no problem (in bulk) though some control transfer do fail. (same with patch and without-patch)

conclusion:
i find that theoritically the problem exists but was unable to reproduce it.
so, until we cannot reproduce the difference, we cannot proceed further.

closed 2015-08-19 by kuldeepdhaka

devanlai, 2016-06-05:
As an additional data point, I've been running into this issue on a custom STM32F042C6 board shortly after enumeration (I can see the first few USB string descriptors are retrieved successfully, but some of the interfaces whose strings are retrieved later fail because of the infinite stall). The workaround in #446 and in the commits that reference this issue seems to resolve the symptoms.