libopencm3/libopencm3 #451: read modify write cycle to be avoided as per doc

kuldeepdhaka, 2015-05-23:
As per documentation, Read modify write for ISTR and Endpoint in usb not recommended.
(p861, p868 RM0091)

fenugrec, 2016-03-30:
I agree, but I doubt that commit you guys added will fix the issue. You're still doing a variant of RMW by doing
1) tempvar = USB_ISTR (or USB_EP)
2) clear one bit
3) write USB_ISTR or USB_EP.

For USB_ISTR, something like this is needed :
` USB_ISTR = ~(USB_ISTR_XYZ); `
(note : this writes 1 in some reserved bits on F0, so it needs to be refined slightly)

For USB_EP, it's more complicated since there are "rw" and "toggle" bits we need to avoid as well as the "rc_w0" flags we're interested in. As it turns out, I think the macros of libopencm3 are **already** doing this properly. The desired operation is thus:
```
/* clearing TX_CTR : */
uint32_t ep_temp = USB_EP(addr) & 0x8F8F & USB_EP_TX_CTR;	//clear "t" bits, keep "rw" bits
ep_temp |= USB_EP_RX_CTR;	//do not clear the other CTR bit (also "rc_w0")
USB_EP(addr) = ep_temp;

/* clearing RX_CTR : */
uint32_t ep_temp = USB_EP(addr) & 0x8F8F & USB_EP_RX_CTR;	//clear "t" bits, keep "rw" bits
ep_temp |= USB_EP_TX_CTR;	//do not clear the other CTR bit (also "rc_w0")
USB_EP(addr) = ep_temp;
```

karlp, 2016-04-02:
what commit are you referring to?

fenugrec, 2016-04-02:
It's a commit kuldeep and gmc added to their repos that claim to fix the issue, but as you pointed out in PR #641 there's no issue after all. Well only the issue of writing 1 to reserved bits in ISTR, not RMW.

kuldeepdhaka, 2016-04-02:
?? @fenugrec please explain what you are "saying".

> It's a commit kuldeep and gmc added to their repos that claim to fix the issue

gmc repo repo referenced the issue earlier, probebly both are my commit.

> but as you pointed out in PR #641 there's no issue after all

did i say anything in #641 ?

fenugrec, 2016-04-02:
@kuldeepdhaka let me rephrase :
- yes, it's your commit in gmc's tree too. No, you didn't say anything in #641 , but karlp did, he pointed out I was wrong in my comments in 641.
- the issue in your commit (kuldeep), is you're now doing an R-M-W for USB_ISTR :
1- line 212 "istr = USB_ISTR;"
2- line 262 " istr &= ~USB_ISTR_SUSP; "
3- line 382 "USB_ISTR = istr;"

Your commit (kuldeep) also slightly changes the handling of USB_EP(), but that change was not necessary since the macros were already OK.

fenugrec, 2016-04-12:
And so, I think this issue should be closed as invalid. There is no RMW in the original code, as was noted in the discussions of PR #641 .