libopencm3/libopencm3 #453: multiple callback register for control and set-config

kuldeepdhaka, 2015-05-26:
it is better to remove MAX_USER_CONTROL_CALLBACK and MAX_USER_SET_CONFIG_CALLBACK
set-config: i have never seen any code that register multiple callback, and a very-very-very narrow use case. it better to multiplex from the callback.
control-callback: let the user handle and compare mask like it want. use the same principal (handled, not-supported ..., not-handled) and let one user callback multiplex it like it want.

https://github.com/libopencm3/libopencm3/blob/master/lib/usb/usb_private.h#L41

kuldeepdhaka, 2015-05-26:
using this method  for control-callback, it give more flexibility to user code.
it can mask, test, and handle it like it want.

kuldeepdhaka, 2015-05-27:
what is so special about set-config and control-transfer callback that we need multiple callback registering inside usb-stack?
we also have reset, suspend, resume, sof, alternate-setting callback.
and same ways, each endpoint callback.
so, why not the same rule for the above callback, what is *so* special in set-config and control-transfer callback?
afaif, since they {set-config, control-transfer} were designed for multiple, we need to keep them multiple and it is our psychological drive that making us add garbage code to continue the tradition.
what i feel is, if we need multiple callback, we make for {reset, suspend, resume, sof, alternate-setting} callback not just limited {set-config, control-transfer}
im not advocating that implementing using linked list or array or function is better.
im saying that if we need multiple callback, every callback deserve the same design.
legitimate usecase of multiple suspend and resume callback: multiple driver (like msc and cdc) can shut down peripheral to reduce power consumption.
reset usecases: most of the set-config usecases applies for reset too. [whenever multiple set-config is required, multiple reset can be required too]
sof: multiple driver could be using sof as a method for synchronization
alternate-setting: multiple driver could register themselves to know when their alternate setting changed by host
so, what i can derive from above, we can do one of the following:
1) keep the confusing and unethical api (only set-config and alternate-setting have multiple callback facility)
2) since every callback deserve equality and have proper usecase, write lots of garbage code for little gain (so that it don't look *hard*)
3) KISS, register one callback and let the user route where ever it want.
i would vote for (3) because it is flexible, less resource demanding (in term of ram), use flash and give user the flexibility to do what ever it want.
my view on (1) would be the crappiest thing to do.
(2) accumulate lots of useless code and library is not minimalist

securez, 2019-11-08:
I see that this issue is open for a long time, I'm making lot of things now with STM32 USB devices, and really understand the frustration with the current USB status of the library. When I try use it to implement multiple devices I hit the wall.

SetConfig / Control callbacks limit : See this problem in other projects that uses the library, all of them copy the offending files into project and patch them, mostly USBD_REQ_NOTSUPP vs USBD_REQ_NEXT_CALLBACK, and another layer to control extra callbacks.

This leads to use the USB library is hard in some real use case scenarios, I also think that a simple approach is the best, less code, less resource usage, and make more ease that people write usb device class support that can be used together without getting a nightmare of patching.

NOTE: I followed the same path, have cdc, msc, usb_standard, etc. with multiple patches applied, but making higher the limit of callbacks, but I'm planning to remove multiple callbacks at all, in my mind I see much less code to maintain.

kuldeepdhaka, 2019-11-08:
@securez Have a look at https://gitlab.com/insane-adding-machines/unicore-mx (you can find above mentioned changes in the repo) and \#unicore-mx IRC channel. You can build on work already done.

karlp, 2019-11-08:
@securez the ticket here is talking about removing the limits, and just having a single one, that the user handles themselves.  You can already do that yourself by just registering a single callback and not using the facility of having more? Am I missing something?  Can you link to projects that are actually having problems? we tend to get ~zero~ feedback on this.