libopencm3/libopencm3 #454: stm32f0 ADC poweroff

trichner, 2015-05-28:
According to the reference manual (DocID018940 Rev 7, page 233) following procedure should be used to disable an ADC.

*Set ADDIS=1 to disable the ADC and put the ADC in power down mode. The ADEN
and ADDIS bits are then automatically cleared by hardware as soon as the ADC is fully
disabled.*

But libopencm3 instead tries to clear the ADEN bit, which is not possible according to the manual (the bit has `rs` access).

This should be the correct way:
```C
    ADC_CR(ADC) |= ADC_CR_ADDIS;
    while (ADC_CR(ADC) & ADC_CR_ADEN);
```
In order to re-calibrate an ADC it is essential to make sure the ADC is fully powered off i.e. `ADEN=0`, so there should be a call like `adc_power_off_wait_finish(...)` since `adc_power_off(...)` is async.

Relevant implementation:
https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/f0/adc.c

kuldeepdhaka, 2015-05-28:
your addition make sense to me.
i had to add external code to wait for adc to enable in my code.

kuldeepdhaka, 2015-05-28:
https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/f0/adc.h#L341
i recommend adding adc_power_off_wait_finish() to continue the tradition.

trichner, 2015-07-13:
Fixed the naming issue on stm32f3, is now also `adc_power_off`

karlp, 2016-03-30:
fixed via landing the f3/f0/l4 adc tree, async on and off and sync on and off for all

closed 2016-03-30 by karlp

trichner, 2016-03-30:
Good work!