libopencm3/libopencm3 #455: STM32F0, ADC poweron only async

trichner, 2015-05-28:
Enabling ADC according to reference manual:
*Follow this procedure to enable the ADC:
1. Set ADEN=1 in the ADC_CR register.
2. Wait until ADRDY=1 in the ADC_ISR register (ADRDY is set after the ADC startup time). This can be handled by interrupt if the interrupt is enabled by setting the ADRDYIE bit in the ADC_IER register.*

while libopencm3 says: (`adc_power_on(...)`)
*If the ADC is in power-down mode then it is powered up. The application needs to wait a time of about 3 microseconds for stabilization before using the ADC. If the ADC is already on this function call will have no effect.*

I suggest an additional function like `adc_power_on_wait_finish(...)` or simply make `adc_power_on(...)` synchronous since I don't see an application where one doesn't care if the ADC is actually ready.

Relevant code:
```C
adc_power_on(ADC);
while (!(ADC_ISR(ADC) & ADC_ISR_ADRDY));
```

kuldeepdhaka, 2015-05-28:
your addition make sense to me.
i had to add external code to wait for adc to enable.

kuldeepdhaka, 2015-05-28:
https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/f0/adc.h#L341
i recommend adding adc_power_on_wait_finish() to continue the tradition.

karlp, 2016-03-30:
fixed via the landing of the f0/f3 adc tree

closed 2016-03-30 by karlp