libopencm3/libopencm3 #458: Missing functions in flash module for STM32L1 family

slintak, 2015-05-29:
I am working on simple IAP bootloader for STM32L1 Nucleo board. This board contains STM32L152RET6.

When I try to compile my code, the GCC and linker returns errors:

```
$ make
main.c:92:9: warning: implicit declaration of function 'flash_program_half_word' [-Wimplicit-function-declaration]
         flash_program_half_word(baseaddr + i, *dat);
         ^
main.c:92: undefined reference to `flash_program_half_word'
collect2: error: ld returned 1 exit status
make: *** [iap_usart.elf] Error 1
```

According to [documentation for STM32L1](http://libopencm3.github.io/docs/latest/stm32l1/html/group__flash__file.html), there should be function `flash_program_half_word()`. But it is missing in header file `libopencm3/stm32/flash.h` and there is no implementation in `libopencm3/stm32/l1/flash.c`.

What am I missing?

karlp, 2015-05-29:
Sounds more like a documentation bug actually :)  If you read the L1 flash programming manual, there's no half word mode, that's something that F1 uses.

slintak, 2015-05-29:
Thank you for quick response. Yes, it looks like bug in documentation.

But is it possible to write into flash memory for STM32L1?

karlp, 2015-05-29:
well, you did find stm32/l1/flash.c, so... you should probably try those :)

karlp, 2016-08-19:
fixed in all 3cc6ecd20330a2219f01a5b42ba5d6f36c232d0b

closed 2016-08-19 by karlp