libopencm3/libopencm3 #459: Add delay functions

rkreis, 2015-05-29:
libopencm3 should have a function that takes a certain number of clock cycles. This could be used to control timing in non-interrupt-driven programs (like `miniblink`) or when very short delays, possibly shorter than the interrupt entry/exit time, are required.

I suggest an implementation based on the `SYSTICK` timer for accuracy and easy portability.  Other options are using a peripheral timer or a carefully timed delay loop.  The API would probably have to be cycle-oriented, because there's not always a way to know the actual clock frequency the program runs at.  There should also be a function to easily initialize the SYSTICK timer, so adding delays is as simple as for example `systick_init_defaults()` followed by `systick_wait_cycles(100)`, which would be implemented along the lines of:
```c
uint32_t start = systick_get_value();
while((systick_get_value() - start) < delay_cycles)
    ;
```

karlp, 2015-05-29:
For posterity, irc discussions generally agreed that systick is a poor choice for this, and that some higher level helper routines for timer periperhals would be more useful, and meet these demands.

BuFran, 2015-05-30:
hi, rkreis. Try this: (I am testing some APIs in my other repo, if it will be so good to add to locm3)

https://github.com/BuFran/hal/blob/master/include/hal/delay.h

rkreis, 2015-05-30:
That's actually quite close to what I'm currently using.  As you say, `_delay_3t` is architecture dependent, so we'd have to test that at least on the bigger Cortexes with branch prediction and whatever other goodies they have.  Additionally, the loop will pause while interrupted, whereas a counter keep running, which would seem more useful.

Summarized and extended:
* SYSTICK: :heavy_plus_sign: available on all Cortex cores but :heavy_minus_sign: will break RTOS and other uses of the timer
* DWT_CYCCNT: :heavy_plus_sign: no configuration needed, doesn't interfere with other uses but :heavy_minus_sign: not available on all MCU (especially M0)
* peripheral timer: there's most likely a spare timer, and you also get a :heavy_plus_sign: prescaler and :heavy_plus_sign: configurable clock input but :heavy_minus_sign: needs an implementation for every MCU
* delay loop: :heavy_plus_sign: uses no peripherals and is :heavy_plus_sign: suitable for very short delays but :heavy_minus_sign: needs an implementation for every core and :heavy_minus_sign: pauses counting when interrupted

I'd say we should offer a simple API for quickly setting up peripheral timers for one-shot use and also provide properly tested delay loops.  That way, we leave the choice up to the user.

stefanct, 2015-12-27:
FWIW I don't think that using peripheral timers are really necessary (of course it would be nice to have)... there is an API for timers anyway - even if it is not generic (I dunno, I am a libopencm3 n00b), but having no simple delay function (that's configurable by a simple CPU frequency define) is really meh.

In most cases the inaccuracies introduced by interrupts are no issue when one uses them. And if they are... one should disable interrupts anyway :)

ChuckM, 2015-12-27:
The on going meta discussion is how much is this library just for configuring Cortex-M chips and their peripherals, and how much is it a set of basic utility functions that let you write your programs. The latter might be something like libc which has functions for number conversion, but the low level LOC3 library has no need to convert numbers.

Delay is a somewhat special case in that some peripheral code it would be helpful (for example there are delays that need to be followed when initializing SD cards) but we seem to not have too many of these high level functions so less need for delay.