libopencm3/libopencm3 #460: Merge of USB host library

amirhammad, 2015-06-01:
I would be happy to get the [usb host library](https://github.com/libusbhost/libusbhost) merged into libopencm3. 

This issue was opened with intent to discuss whether and possibly when and how could it be done.

ChuckM, 2015-06-01:
It is an interesting question Amir, I've been developing a bunch of higher level APIs for things like SDIO and graphics which I've been building into a monolithic mini sort of OS for my projects. I've added yours to get some experience with it (most of my work is with the STM32F4 series anyway) and it lined up with my future need to add a mouse and keyboard input.

But the challenge with adding it to libopencm3 directly is that, not all that many variants have USB-OTG peripherals. The Tiva C does, the STM32F4 does, and I believe the SAM4 does but those are all Cortex-M4 variants. For one offs it makes sense to just link in an additional library.

amirhammad, 2015-06-01:
@ChuckM The library was meant to be as multiplatform as it could be, but as of this time, only the low level driver for stm32f4 was written. I think, developing usb device drivers (for example: mouse, keyboard drivers) is easier when it is independent of the underlying hardware (OTG in case of stm32f4). However, USB-OTG is present in stm32f4 f2 and f1 (f105 or something), device drivers could be written independently of the low level implementation. 

The case could be: The programmer wants to use some special platform for which low-level driver does not exist yet. He, then, instead of writing his own low+high-level driver, could write only low level driver and then enjoy all the features device drivers contained in libopencm3 offer.

amirhammad, 2016-07-07:
I think this issue is no more considered and so I close it.

closed 2016-07-07 by amirhammad

jhgorse, 2021-06-19:
@ChuckM @karlp @gsmcmullin would it make sense to make a note about the scope of opencm3 at the top level so that folks can know that it is not intended for USB Host stack or other things which negatively affect interoperability?

Also, it's been a while. I am glad to see this is still moving forward. I read through some of the USB device code stack and it looks quite nice. =)

Cheers,
Joe

karlp, 2021-06-21:
We've got nothing against bringing host support in, but it's not going to happen via some random issue pointing at some external code.  :)  I'm certainly not going out to external repos and restructuring them to bring them in.