libopencm3/libopencm3 #468: Linker files are missing in install/build

kuldeepdhaka, 2015-07-02:
When installing/building code, all linkers should be copied to /lib, not just the generic one.
for example: in stm32/l1, their are
libopencm3_stm32l1.ld  stm32l15xx8.ld  stm32l15xxc.ld
stm32l15xx6.ld	       stm32l15xxb.ld  stm32l15xxd.ld
but only libopencm3_stm32l1.ld is being copied.

reported by smithmike.

karlp, 2015-07-02:
we sort of stopped on the path there, I was adding full sets of xxY.ld for l1 series, 
               and that's what bufran's scripts in ld are fo, to generate the full set, without 
               having to have lots of copies of files

For instance, even though there's a set of L15xxY files, there's nothign for any of the other familes beyond the basic family template.  (And even those are actually virtually identical) The hope was to migrate away from having all these committed ~identical~ files.  But until we have, it's a fair call to install them :)

karlp, 2016-08-18:
We simply flat out dropped the install target in 7c1cd2e5cd3f3d1193ee375b46cfd8afaccb89f7 so this is no longer relevant.  Long live the linker generator!

closed 2016-08-18 by karlp