libopencm3/libopencm3 #469: Why are shifts defined like #define FMC_SDCR_SDCLK_SHIFT (1 << 10)

mox-mox, 2015-07-02:
There are a number of shift definitions like the one in the title, for example in libopencm3/include/libopencm3/stm32/f4/fmc.h, and I would like to know why they are not set like
#define FMC_SDCR_SDCLK_SHIFT 10
That way, the shift operator could be used to actually shift values into the registers like 

FMC_SDCR |= some_val << FMC_SDCR_SDCLK_SHIFT;
instead of things like this
	result |= ((t->trp - 1) & 0xf) * FMC_SDTR_TRP_SHIFT; // taken from libopencm3/lib/stm32/f4/fmc.c

And the set of definitions would just be

/* SDCLK: SDRAM Clock Configuration */
#define FMC_SDCR_SDCLK_SHIFT		10
#define FMC_SDCR_SDCLK_MASK		(3 << FMC_SDCR_SDCLK_SHIFT)
#define FMC_SDCR_SDCLK_DISABLE	(0 << FMC_SDCR_SDCLK_SHIFT)
#define FMC_SDCR_SDCLK_2HCLK		(2 << FMC_SDCR_SDCLK_SHIFT)
#define FMC_SDCR_SDCLK_3HCLK		(3 << FMC_SDCR_SDCLK_SHIFT)

instead of

/* SDCLK: SDRAM Clock Configuration */
#define FMC_SDCR_SDCLK_SHIFT		(1 << 10)
#define FMC_SDCR_SDCLK_MASK		(3 << 10)
#define FMC_SDCR_SDCLK_DISABLE		(0)
#define FMC_SDCR_SDCLK_2HCLK		(2 << 10)
#define FMC_SDCR_SDCLK_3HCLK		(3 << 10)

So, what advantage am I missing here?

ChuckM, 2015-07-02:
Its a style thing, as you can see in the library it goes back and forth between styles. Often depending on the author and sometimes depending on when the author wrote something or whether or not it was new code or additions to existing code. I'm all in favor of having a big check-in that fixes all of them to be the same.

mox-mox, 2015-07-03:
Ok, then I might just do that at least for the STM32F4 part of the lib.

mox-mox, 2015-07-03:
Ok, turns out a quick grep -iR "define.*shift.*<<" shows this style is only used within fmc.h.
I have changed it there. As far, as I know, no further parts of the lib and libopencm3-examples are affected.
I do however plan to do some more programming for the fmc. Should I create a pull request now, or should I wait until I have done more?

karlp, 2015-07-03:
that particular style in FMC does indeed look very unusual.  there's been some back and forth on bit definitions, but particularly, naming a shift with a << does seem rather odd.  The two ways I'd accept for a change to the FMC would be so it ends up like....

    #define option_disable 0
    #define option_1 1
    #define option_2 2
    #define option_shift 10
    #define option_mask 0x3

or

    #define option_disable (0 << 10)
    #define option_1 (1 << 10)
    #define option_2 (2 << 10)

I personally vastly prefer the former, but both styles exist.   I feel the former makes it easier to do switches on logical values but it doesn't make a huge difference

ChuckM, 2015-07-03:
Clearly this include file is utter crap, who wrote it? To be honest I don't know what I was thinking and it was fairly early in my use/adoption of LOC3 but that isn't much of an excuse. I don't think there are very many uses of the code though. SO we can get away with changing it.

karlp, 2015-07-29:
@mox-mox, please make a nice small pull with just the removal of those strange shifts first.  It will make it easier to review later code if you havemore FSMC changes coming.

karlp, 2015-10-06:
Shifts are fixed via 17bc66c46bf07ea0d7c88b3769b5ea90de6aace2

closed 2015-10-06 by karlp