libopencm3/libopencm3 #476: usb_f103.c : bad ep_rx_bufsize calculation

fenugrec, 2015-07-19:
in lib/usb/usb_f103.c ::usb_set_ep_rx_bufsize() ( https://github.com/libopencm3/libopencm3/blob/master/lib/usb/usb_f103.c#L88 ), the code will produce invalid values :
- sizes <= 62 are rounded up (ok) but not shifted >>1
- sizes > 62 are rounded down (inconsistent) and not re-adjusted
- (minor) : size==0 is not checked; but is not allowed for NUM_BLOCK

fenugrec, 2015-07-19:
I'm having trouble making a PR but the correct function would be:

```
/*
* Valid for at least F102, F103, and F0 devices
*/
static void usb_set_ep_rx_bufsize(usbd_device *dev, uint8_t ep, uint32_t size)
{
	(void)dev;
	if (size > 62) {
		if (size & 0x1f) {
			size += 32;
		}
		size >>= 5;
		size -= 1;
		size |= 0x20;	//set BSIZE bit
	} else {
		if (size & 1) {
			size ++;
		}
		size >>= 1;
	}
	USB_SET_EP_RX_COUNT(ep, size << 10);
}
```

kuldeepdhaka, 2015-10-02:
```
#include <stdio.h>
#include <stdint.h>
#include <inttypes.h>

static uint16_t calc_current(uint32_t size)
{
	if (size > 62) {
		if (size & 0x1f) {
			size -= 32;
		}
		return (size << 5) | 0x8000;
	} else {
		if (size & 1) {
			size++;
		}
		return size << 10;
	}
}

static uint16_t calc_new(uint32_t size)
{
	if (size > 62) {
		size += 31; //round up to nearest mult. of 32
		size >>= 5;
		size -= 1;
		size |= 0x20;   //set BL_SIZE bit
	} else {
		if (size == 0) {        // illegal...
			size = 2;
		}
		size++;
		size >>= 1;     //round up and divide by 2
	}

	return size << 10;
}

#define BL_SIZE_SHIFT 15
#define NUM_BLOCK_SHIFT 10
#define NUM_BLOCK_MASK (0x1F << NUM_BLOCK_SHIFT)

static void test(uint16_t result, uint16_t size, uint16_t (*calc)(uint32_t))
{
	uint16_t actual_result = calc(size);
	if (result != actual_result) {
		printf("FAIL expected-result=%"PRIx16", "
				"actual-result=%"PRIx16", "
				"size=%"PRIu16"\n",
				result, actual_result, size);
	} else {
		printf("PASS result=%"PRIx16", "
				"size=%"PRIu16"\n", result, size);
	}
}

void test_func(const char *name, uint16_t (*calc)(uint32_t))
{
	unsigned i;

	printf("============= %s =========\n", name);

	/* for size=2..62 */
	for (i = 1; i < 31; i++) {
		test((0 << BL_SIZE_SHIFT) | (i << NUM_BLOCK_SHIFT), i * 2, calc);
	}

	/* for size=32, 64, 96 ... 992 */
	for (i = 0; i < 31; i++) {
		test((1 << BL_SIZE_SHIFT) | (i << NUM_BLOCK_SHIFT), (i + 1) * 32, calc);
[... 112 lines cut ...]
PASS result=a400, size=320
PASS result=a800, size=352
PASS result=ac00, size=384
PASS result=b000, size=416
PASS result=b400, size=448
PASS result=b800, size=480
PASS result=bc00, size=512
PASS result=c000, size=544
PASS result=c400, size=576
PASS result=c800, size=608
PASS result=cc00, size=640
PASS result=d000, size=672
PASS result=d400, size=704
PASS result=d800, size=736
PASS result=dc00, size=768
PASS result=e000, size=800
PASS result=e400, size=832
PASS result=e800, size=864
PASS result=ec00, size=896
PASS result=f000, size=928
PASS result=f400, size=960
PASS result=f800, size=992

so, this proves that this PR is useful and fix a actual problem.
@fenugrec good work!

fenugrec, 2016-10-04:
Rebased the commit and updated PR name.

karlp, 2018-07-25:
I'm wondering whether it wouldn't be best to change the API here to to return teh number of bytes actually allocated?

fenugrec, 2018-07-26:
May I direct your attention to line 47 of the ancient scrolls of PR 880 !
https://github.com/libopencm3/libopencm3/pull/880/commits/4d8076ebab0145ad753497f2efce0b95cb00d724#diff-8ac26e98bbf7121474dfa16940d813f0R47

karlp, 2018-07-26:
yeah, I found it myself later looking at different chunks of this.

closed 2018-08-17 by karlp