libopencm3/libopencm3 #491: usb:STM32F103CBT6 not working

rgwan, 2015-07-27:
I have compiled the example usb-hid, and insert to my computer, the output from the kernel is like this:
[18706.705864] usb 1-1.3: new full-speed USB device number 6 using sw-ehci
[18711.825890] usb 1-1.3: unable to read config index 0 descriptor/start: -110
[18711.836629] usb 1-1.3: chopping to 0 config(s)
[18716.847339] usb 1-1.3: string descriptor 0 read error: -32
[18716.857935] usb 1-1.3: no configuration chosen from 0 choices

I don't know how to solve it...Please help me to fix it, thanks!

karlp, 2015-07-27:
as described on irc, please provide which actual example this was, and what board you are working with.

rgwan, 2015-07-27:
I use a board made by myself, USB part 's wiring just like h103.
I used H103/USBHID project.

rgwan, 2015-07-28:
I  have print the status on HardFault_ISR. It surprised me:
SP=0x020027d8 PC(before)=0x0800172c
Here 's the info from the map.
                0x080016d4       0x28 ../../../../../../libopencm3//lib/libopencm3_stm32f1.a(usb_standard.o)
 .rodata.CSWTCH.26
                0x080016fc       0x30 ../../../../../../libopencm3//lib/libopencm3_stm32f1.a(usb_standard.o)
                0x0800172c                . = ALIGN (0x4)

.glue_7         0x0800172c        0x0
 .glue_7        0x0800172c        0x0 linker stubs
what is glue_7?

rgwan, 2015-07-28:
Use new compiler fixed it.
DON'T USE CROSSTOOL_NG to build toolchain! it waste my two days to debug!

closed 2015-07-28 by rgwan