libopencm3/libopencm3 #493: STM32F3 ADC: wrong operation on ADC_SR flags

fenugrec, 2015-07-27:
A few functions in lib/stm32/f3/adc.c  :
- adc_clear_overrun_flag()
- adc_eoc_after_each()
- adc_eoc_after_group()

do not use the ADC_SR register properly : to clear a given flag, a 1 must be written to the appropriate bit. The current code does a read-modify-write operation (useless) and attempts to write 0 to the specified bit (no effect) !

karlp, 2015-07-28:
Yup.  but it's not the ADC_SR, it's ADC_ISR :)  Also, this file is only appropriate for the f30x and f31x devices.  The f37x devices have the other adc peripheral, with an d_SR register, and rc_w0 bits, instead of r_w1 bits.

fenugrec, 2015-07-28:
Good points; I should've specified I only checked in RM0365 r4  (F302x only); those have r_w1 bits.

karlp, 2015-07-29:
clear_overrun_flag is easy to fix. but the eoc_after_each and eoc_after_group I don't even think are doing the right thing at all!

karlp, 2015-07-29:
Fixed via a set of commits just merged.  OVR is fixed, and the other ones have been removed and the "right" thing done instead.

closed 2015-07-29 by karlp