libopencm3/libopencm3 #500: usbd function naming convention

kuldeepdhaka, 2015-07-30:
odd but some function 

use usbd_set_*()
 - usbd_register_set_config_callback
 - usbd_register_set_altsetting_callback

some use usbd_*_set()
 - usbd_ep_stall_set
 - usbd_ep_stall_get
 - usbd_ep_nak_set

karlp, 2015-07-30:
I know what you're getting at, but this is a general task for the entire library to try and be more consistent.  Most of the library seems to use peripheral_set_something and peripheral_get_something, but blah_clear_ isn't so consistent.

Would you propose

    usbd_set_ep_stall
    usbd_get_ep_stall
    usbd_set_ep_nak

?  that seems generally reasonable, but I feel this is just noise for right now, we've got a lot of other usb to try and merge and test.

kuldeepdhaka, 2015-07-30:
@karlp yea, we can fix this later, but still a task (low priority) :)