libopencm3/libopencm3 #513: Including libopencm3/stm32/pwr.h on STM32 F2 architecture fails to import correctly.

tdicola, 2015-08-10:
I'm working on a STM32F2 project and noticed when I include pwr.h I get a warning that I shouldn't explicitly include pwr_common_all.h (I'm not) and none of the power defines are set correctly.

For example in my program I have this at the top:
```
#include <libopencm3/stm32/pwr.h>
```
(I have no other pwr related defines in my code)

When I compile I see this warning:
```
In file included from ../../libopencm3/include/libopencm3/stm32/f2/pwr.h:36:0,
                 from ../../libopencm3/include/libopencm3/stm32/pwr.h:28,
                 from uart.c:18:
../../libopencm3/include/libopencm3/stm32/common/pwr_common_all.h:129:2: warning: #warning "pwr_common_all.h should not be included explicitly, only via pwr.h" [-Wcpp]
 #warning "pwr_common_all.h should not be included explicitly, only via pwr.h"
```
And I get failures that power related defines I need to use like PWR_CSR, etc. aren't set.

Looking more into the code it looks like the STM32 F2 pwr.h has a small issue with how it sets defines.  From stm32/common/pwr_common_all.h it seems to check for a LIBOPENCM3_PWR_H define here: https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/common/pwr_common_all.h#L31  Looking at other STM32 processors like the F4 they set that define explicitly before including pwr_common_all.h, like here: https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/f4/pwr.h#L34-L37  However STM32 F2's pwr.h differs from all the others and doesn't set that define, instead it sets LIBOPENCM3_PWR_F2_H: https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/f2/pwr.h#L33-L36  Since LIBOPENCM3_PWR_H isn't set when pwr_common_all.h is imported it skips defining all the important power defines and throws the warning.

This looks like a bug to me but I just wanted to check am I doing anything wrong or unexpected?

tdicola, 2015-08-10:
Also as a workaround for anyone else that might hit this issue, you can manually define LIBOPENCM3_PWR_H before including libopencm3/stm32/pwr.h and everything works fine:
```
#define LIBOPENCM3_PWR_H
#include <libopencm3/stm32/pwr.h>
```
It's a bit hacky but at least works around the issue for now.

ksarkies, 2015-08-10:
> b.com/li
> bopencm3/libopencm3/blob/master/include/libopencm3/stm32/f2/pwr.h#L33-L36

Hi Tony

Yes there is a bug for F2 and F3. The pwr_common.h file looks for a 
define #ifndef LIBOPENCM3_PWR_H that should be in pwr.h for each given 
family, and complains if not found. In fact it will exclude all the 
common defines which is why you didn't get the pwr defines correctly. 
For F2 it has #ifndef LIBOPENCM3_PWR_F2_H. For the moment you could use 
your suggested workaround or just change the define to remove the _F2 
and I'll put up a PR to fix it.

Ken

karlp, 2015-08-10:
thanks for the report and diagnosis! much appreciated.  Fixed via ksarkie's pull in 4d31528

closed 2015-08-10 by karlp

tdicola, 2015-08-13:
Awesome thanks for the quick fix!