libopencm3/libopencm3 #516: Stack pointer is not always 8-byte aligned during ISRs

forrestv, 2015-08-13:
Alignment of the stack pointer along 8-byte boundaries is required by the ARM ABI during function calls[1]. As was determined from a conversation on #libopencm3, this can be violated for interrupt handlers when the STKALIGN bit in SCB_CCR isn't set. It is set by default for Cortex revision r2p0, but not r1p1.

As the libopencm3 examples don't set STKALIGN manually or use \_\_attribute\_\_((interrupt)) [2] on ISRs, users that follow the examples will find that ISRs that use varargs (potentially among other things) will mysteriously fail occasionally, when the code running before the interrupt happened had the stack not 8-byte aligned (as is allowed).

References:
1. Last section of ARM IHI 0046B
2. Last line of the interrupt section of https://gcc.gnu.org/onlinedocs/gcc/ARM-Function-Attributes.html

forrestv, 2015-08-13:
See pull request #517 for fix.

karlp, 2015-08-14:
closed via 4a21730

closed 2015-08-14 by karlp