libopencm3/libopencm3 #520: usb data schedule api for reading and writing to endpoint

kuldeepdhaka, 2015-08-21:
conversation with zyp (the brain child) of the design.

tl;dr; see at the end

\<zyp> as of current code, all configured out endpoints are always enabled unless explicitly naked, and once a packet is received, you have to call ep_read()
\<kuldeepdhaka> zyp, good thing, if someone is configuring ep, they are expecting data. right?
\<zyp> IMO a better API would have all endpoinds naked by default, and something like schedule_read(ep, buf, size) which then disables nak and notifies user code when the buffer is filled
\<zyp> that way, you'll never end up in the situation where you receive data in the peripheral buffer, but have no user buffer to move it to
\<kuldeepdhaka> zyp, i feel the opposite. but if there is some usb perip (like efm32) with integerated DMA to write data to any address, yes then it could be useful
\<zyp> it doesn't have anything to do with DMA or not, those are implementation details
<kuldeepdhaka> current design is simple in term of api (and is designed around usb perip that have inbuild memory)
\<zyp> simple? sure
\<zyp> but explicitly maintaining flow controll is a hassle with a bunch of pitfalls
\<kuldeepdhaka> yes, ep_read is quite straightforward
\<kuldeepdhaka> with schedule_read() , application code get more complexity. states are added
\<zyp> on the contrary, it gets less complex
\<zyp> because application code doesn't need to maintain flow control anymore, it's implicit
\<kuldeepdhaka> zyp, maybe we can get "schedule_read" and "schedule_write" in future.
\<zyp> write logic wouldn't really change
\<kuldeepdhaka> current design, assume independent memory for usb periph
\<kuldeepdhaka> zyp, no, schedule_write could internally be used to prevent copying over to pma
\<zyp> current design was designed around stm32f1's usb peripheral, and is sort of a poor fit even for dwc_otg in stm32f4
\<zyp> just think back to this whole fifo clearing issue
\<kuldeepdhaka> zyp, dwc_otg has inbuild memory. (btw im not much aware of the fifo design and working)
\<kuldeepdhaka> (atleast for efm32, stm32)
\<zyp> kuldeepdhaka, there's a huge difference between a common rx fifo, and individual rx buffers for each endpoint
\<kuldeepdhaka> zyp, im not aware of that difference, not much into design.
\<zyp> on stm32f1, there's no problem leaving packets sitting in the peripheral buffer, a packet for one endpoints doesn't block reception of the other
\<zyp> on dwc_otg, there's a common rx fifo for all endpoints, so once a packet is received on any endpoint, it must be read out before any other endpoints can receive anything
\<kuldeepdhaka> (stm32f1  which? otg or device)
\<zyp> device.
\<kuldeepdhaka> zyp, im aware of the device deisgn
\<zyp> and this led to the requirement that ep_read() must be called in the rx callback, otherwise the data would get dropped to make room for the next packet
\<zyp> which is kind of silly, here it would be much more sane to have a buffer ready from schedule_read() that you could just drop the data into and continue
\<kuldeepdhaka> zyp, (that why finite loop problem on device)
\<zyp> among other stuff
\<zyp> my stack also works like this, so it's possible to do it like this, just not optimal
\<zyp> and then there's the whole matter of flow control
\<kuldeepdhaka> zyp, with current design, if application code want to drop data (very very unlikely) then automatically copying is just wastage of cpu cycles  [just a minor point, nothing serious]
\<zyp> kuldeepdhaka, exactly
\<zyp> «very very unlikely»
\<zyp> with my proposed model, the application code wouldn't schedule a transfer in the first place if it wanted to drop data
\<kuldeepdhaka> zyp, if buffer is there, accept data or else nak
\<kuldeepdhaka> if scheduled buffer is there
\<zyp> exactly
\<zyp> and once you're on that model, user code can schedule transfers larger than a single packet as well
\<kuldeepdhaka> (for OUT this make difference, but for IN nothing different)
\<zyp> so user code doesn't have to handle every single packet manually
\<zyp> yes, that's what I said before
\<zyp> 20:35:00 < zyp> write logic wouldn't really change
\<kuldeepdhaka> zyp, yes good point. efm32 usb stack can schedule multiple packets.  (possbly stm32 has simiar feature but upper limit to 3?)  {need to confirm the datasheet for both}
\<kuldeepdhaka> *efm32 usb perip
\<zyp> it's a feature of the dwc_otg core
\<kuldeepdhaka> zyp, (afair, dma feature is not there in stm32 but multiple packet from fifo can be scheduled)
\<zyp> OTG_HS in has DMA, OTG_FS does not
\<kuldeepdhaka> zyp, oh
\<kuldeepdhaka> so, your design of "schedule_read" "schedule_write" is something that make sense as well as effecient as well as more abstract
\<kuldeepdhaka> more abstract design
\<kuldeepdhaka> (abstract design because it do not require usb perip to have inbuild memory while retaning compatible with with-memory periph)
\<zyp> yes, I imagine it would simplify user code greatly, without really complicating stack code much
\<kuldeepdhaka> zyp, "simplify user code greatly" i do not know, still need to think on that.  (but yea i have a array of values that need to be chunked and send on tx-success callback)
\<kuldeepdhaka> (in current design) , with new design, i can schedule the whole thing in once
\<zyp> have you ever done anything about NAK management in user code?
\<kuldeepdhaka> "done anything about NAK"   you mean set nak or clear nak?
\<kuldeepdhaka> but with the scheduled read/write every code out there need to be changed.  now stack allocation cannot be used
\<zyp> sure it can, as long as the function scheduling doesn't return before the transfer is completed
\<kuldeepdhaka> zyp, blocking read/write?
\<zyp> in my stack, all usb processing is performed by calling usb.process() regularly, IIRC libopencm3 does something similar
\<zyp> so assuming the application doesn't do something else, it could schedule a transfer, block until it finishes while calling usb.process() or equivalent
<kuldeepdhaka> poll()
\<zyp> or in a threaded setup, you could block the current thread while processing happens in a different thread
\<kuldeepdhaka> zyp, i would not recommend that just because of compatibility
\<zyp> stack allocation in the old model isn't that useful anyway, since you have to read in the callback
\<kuldeepdhaka> zyp, threaded  no no, this is a library not an rtos
\<zyp> so you have to allocate on the callback stack, which means you need to process it immediately before returning from the callback
\<kuldeepdhaka> zyp, i would recommend that schedule_read and scedule_write function with similar properties, not with a cocktail of current and schedule functionality
\<kuldeepdhaka> that will add lots of confusion
\<kuldeepdhaka> if we say in abstract, the peripherial is not required to have inbuild memory, then the same applies to out as well as in ep
\<zyp> true

a new api so, that instead of using "ep_read" and "ep_write", "schedule_read" and "schedule_write" function are accepted. the usb periph read/write to the passed memory and when it is done, inform using callback (until callback the passed memory need to be constant).
schedule_read and schedule_write will will accept multiple packets, then application code do not have to perform chunking on each callback.
this design will allow periph with inbuild dma to perform better and less headache to user.
the design also do not require usb periph to have inbuild memory.
(the new api can coexists with old api)

ChuckM, 2016-08-28:
Hard to follow what is in your head but if I understood the conversation, there is a thread of control which is periodically flushing out the FIFO into local memory, when it lands in local memory it notifies the user thread with a callback. Is that accurate?

kuldeepdhaka, 2016-08-29:
@ChuckM atm api similar to usb host.