libopencm3/libopencm3 #531: STM32: USB/OTG: Problems when interrupted while filling FIFO

schodet, 2015-09-19:
When calling usbd_poll in interrupt, a task which calls usbd_ep_write_packet can be interrupted by the OTG interrupt. The problem is that the OTG device does not support any other access while filling the FIFO.

Seen in Chibios: http://sourceforge.net/p/chibios/svn/HEAD/tree/trunk/os/hal/ports/STM32/LLD/OTGv1/usb_lld.h#l128 :

> Because an undocumented silicon behavior the operation of copying a packet
> into a TXFIFO must not be interrupted by any other operation on the OTG
> peripheral.

I added IRQ lock in my code and this solves USB being stuck after some seconds.

karlp, 2015-09-19:
Is this ticket to track adding some documentation? Or do you want irq locking provided internally? or what are you suggesting here?

schodet, 2015-09-20:
Some kind of IRQ locking would be great, but I understand that only a fraction of users will want this. May be disabling only USB interrupts?

karlp, 2015-09-21:
Could you just setup interrupt priorities in your own code? or do the locking in your own IRQ handler?  I'm not sure I see it as being at all correct to add irq locking into a routine named _poll.  By all means it's something that can be added to the documentation though.

schodet, 2015-09-23:
Locking must not be done in poll, but in write_packet while filling the FIFO.
If this is documented ("write_packet must not be interrupted by any other USB access"), this means that this will be imposed on every drivers while it is only necessary for OTG. That does hovewer seem a fair limitation.

gordon-quad, 2016-01-09:
Here is the dirty patch I made for libopencm3 which uses ChibiOS method of solving this.
Any suggestions how I can prettify it to fit libopencm3?
http://ix.io/nhK