libopencm3/libopencm3 #533: Nominal STM32 F0 watchdog clock is 40kHz, not 32kHz according to datasheet

onnokort, 2015-09-27:
Hi,

According to http://www.st.com/st-web-ui/static/active/en/resource/technical/document/reference_manual/DM00031936.pdf, page 553, the IWDG watchdog in STM32F0 parts has a nominal watchdog LSI clock of 40kHz.

Also, my STM32F0DISCOVERY test board's IWDG nominal frequency is close to 40kHz, but not 32kHz.

IMO, there should be either one of two things:

a) a prominent note in the libopencm3 code that iwdg_set_period_ms(..) assumes a 32kHz IWDG LSI clk
or 
b) a separate implementation of iwdg_set_period_ms(..), that calculates the watchdog settings from the nominal 40kHz clock.

Thoughts?

karlp, 2015-09-28:
Blame ST for making LSI 32k on some parts and 40k on other parts.  You're right though, this should be addressed in some way.  Suggestions welcome.

onnokort, 2015-09-28:
I suggest splitting it up and having a function that divides by 40 for any STM32F0 part as a start and copy & paste the old implementation for all others & put in a warning that the LSI might be at 40kHz. I volunteer to make a patch for that if that has a chance of being accepted.

Am I right that the linked PDF describes all STM32F0's? I am a bit confused with all the various documents from ST...

karlp, 2015-09-29:
it would be perfectly valid, just depends how you do it :)  Please bear in mind, the patch queue is fairly long right now, and the current priority is merging f0 and l0 usb support.

If you want to be super awesome, please make sure the commits have good clear summary lines, and ideally, for things like this, some example device code that should run on both old/new and can show the failure.  It makes things much easier for people to verify the changes.  Finally, please make sure you check all the STM32 reference manuals to get the right values for the right targes

karlp, 2017-10-26:
I propose easiest for this is to simply move the #define LSI_FREQUENCY 32000
out of the iwdg.c file and into the iwdg.h headers for each target.  Should be pretty easy

yeyue1, 2026-11-16:
Working on this.

The approach follows the suggestion above: move the nominal LSI frequency into the device headers and let `iwdg_set_period_ms()` derive the counter value from it instead of the hard coded `period << 3`.

I will keep the change to the F0 and F1 families, whose 40kHz LSI is documented (RM0091 / RM0008), and leave the other families on the current 32kHz value so the patch stays reviewable; the value is then overridable per target or per application, so the remaining families can be corrected one at a time.

Local verification: all twelve STM32 families still build with arm-none-eabi-gcc 14.2.1 (no warnings), the generated code for the 32kHz families is unchanged, and the function itself was exercised on the host over 1ms..32760ms with the programmed PR/RLR converted back through the ST formula. PR follows shortly.