libopencm3/libopencm3 #537: TX buffer size is not a multiple of 2 in usb_f103 usb

kuldeepdhaka, 2015-10-02:
https://github.com/libopencm3/libopencm3/blob/master/lib/usb/usb_f103.c#L123
2 byte alignment is not done, it will cause problem to future endpoint setup

karlp, 2015-10-02:
Is this a new issue in anyway? is this not the same as https://github.com/libopencm3/libopencm3/issues/461 ?

kuldeepdhaka, 2015-10-02:
"USB_ADDRn_TX  
Bit 0 Must always be written as ‘0 since packet memory is half-word wide and all packet buffers
must be half-word aligned." [872/988] [DocID018940 Rev 6] [RM0091]

if the max_size is not a multiple of 2 it could result in known behaviour

```
if (dir || (addr == 0)) {
       /* convert max_size multiple of 2 (up) */
	if (max_size & 1) {
		max_size += 1;
	}
       ......
       .....
}
```

karlp, 2015-10-02:
thanks for the extra detail, much appreciated.

fenugrec, 2016-10-06:
Update : since the st_usbfs code merge, this is now in st_usbfs_ep_setup() :
https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/common/st_usbfs_core.c#L62

karlp, 2018-08-20:
I've tried this a lot more, and while I do totally agree that according to the ref man this needs to be "ensured" I'm not at all convinced that you can actually do anything useful with this.  You need to odd byte sized IN endpoints just for _starters_ and then somehow expect those to actually work.

I can't even get the pure source/sink functions to work with odd sized endpoints and mismatched in/out sizes, let alone loopback and witness a buffer trampling.  Unless someone can demonstrate this, I'm closing this as irrelevant.

closed 2018-08-20 by karlp