libopencm3/libopencm3 #547: RFC: ADC power on/off synchronous

karlp, 2015-10-17:
I'm working on merging the changes in #454 #455 #456 #457 #471 and #534 

We currently have some synchronous API commands, (Defined as anything with a while loop checking a flag before it returns)
```
uart_send_blocking()
eth_enable_checksum_offload()
eth_smi_transact()
eth_desc_init()
i2c0_xxx() // lpc43xx
uart_read() // lpc43xx, not conflicts with stm32
adc_start_conversion_regular/injected
crypto_set_iv()
iwdg_set_period_ms()
rtc_set_wakeup_time()
spi_send() && spi_receive() // not conflict with uart!
spi_xfer()
spi_send8()
adc_reset_calibration()
rtc_awake_from_off()
write_i2c() // f3, bad naming too!
sdram_command()
lm4f_usbd_init()
```

and some asynchronous commands.  (anything that just sets a register and returns)
```
uart_send()
uart_recv()
spi_write()
rcc_osc_on()  // << this is the most common one
// Some others, but not as many as you'd think
```

and some async helpers. (Things that don't do anything except wait)
```
uart_wait_send_ready()
uart_wait_recv_ready()
crypto_wait_busy()
uart_wait_send_ready()
rcc_wait_for_osc_ready()
rcc_wait_for_pll_ready()
flash_wait_for_last_operation()
lcd_wait_for_xxx()
```

Not a comprehensive list, but the point is somewhat clear.  Most of the higher level helper routines are actually synchronous.  In a few cases there are async versions, and for those, there's a clear naming convention on the helpers.  `xxx_wait_yyyyy`  There's almost _no_ convention on a function call though as to whether it will be synch or not.  I'm currently planning on making the basic adc_power_on/off calls synchronous, and _maybe_ adding xxxx_async() calls that do the same thing, but skip the waiting, and force you to use the corresponding xxx_wait_yyyyy calls.

fenugrec, 2015-10-17:
(summarizing a short IRC discussion )
IMO the x_wait_y functions are less important than functions to check the result of an async operation (say, `bool x_is_y()` ). Then, the x_wait_y() functions can have simple implementations of the form `while (! x_is_y()) {}` . Otherwise, there is no point in using an async function if the only way to check the result is to call a blocking function p-)

karlp, 2016-03-30:
this has largely landed for adc, both forms are available

closed 2016-03-30 by karlp