libopencm3/libopencm3 #548: F3 ADC definitions renamed

karlp, 2015-10-17:
I'm planning on renaming a pile of definitions for the F3 ADC peripheral (F30x style) to use naming more in line with f0/l0/l4, that share the same peripheral.  It's not much, just 
```
- ADC_CFGR_XXX
+ ADC_CFGR1_XXX
```

This means we can use the same code for both of them.

karlp, 2015-10-30:
After conversations, I'm going to attempt to add the ADC_CFGR defintions for f3 only, as aliases for ADC_CFGR1

esden, 2015-12-27:
@karlp this patch looks good to me. Should we make the "old macros" generate compiler warnings that the CFGR style is legacy and might be removed in a future release?

esden, 2015-12-27:
Ahh I see, this is part of a series, that is a draft pull request #555

karlp, 2016-03-30:
this tree has landed

closed 2016-03-30 by karlp