libopencm3/libopencm3 #558: arm-none-eabi-gcc fails on stm32/f7 lib folder, project cannot be built

MightyPork, 2015-11-02:
```
Assembler messages:
Error: unknown cpu `cortex-m7'
Error: unrecognized option -mcpu=cortex-m7
../../Makefile.include:44: recipe for target 'gpio.o' failed
make[2]: *** [gpio.o] Error 1
Makefile:67: recipe for target 'lib/stm32/f7' failed
make[1]: *** [lib/stm32/f7] Error 2
Makefile:55: recipe for target 'lib' failed
make: *** [lib] Error 2
```

Temporarily solved by removing the folder, but it's not ideal if you want to use the submodule setup.

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

kuldeepdhaka, 2015-11-02:
same issue for me. (im on the same OS and compiler version)
can also be solved (temp) by removing "stm32/f7" from TARGETS in top Makefile

karlp, 2015-11-02:
for real? how do you get this?

```
  CC      sync.c
  CC      dwt.c
  AR      libopencm3_stm32f4.a
  BUILD   lib/stm32/f7
Makefile:60: Your toolchain doesn't support -mcpu=cortex-m7.
Please use gcc-arm-embedded 4.8 2014q3 or newer. Skipping this sub-library.
  BUILD   lib/stm32/l0
  CP      libopencm3_stm32l0.ld
  CC      gpio.c
  CC      rcc.c
```

Are you sure you're on the same tree? the lib/stm32/f7/Makefile very explicitly skips things...

kuldeepdhaka, 2015-11-02:
`arm-none-eabi-gcc -fsyntax-only -mcpu=cortex-m7 -mthumb -xc /dev/null 2>&1`
prints nothing for me

makefile expecting gcc to print something that detect support?

karlp, 2015-11-02:
it's not meant to print anything, it's meant to have a return code.

karlp, 2015-11-02:
should be something like this
```
karlp@cuth:~/src/libopencm3 (master)$ ~/tools/gcc-arm-none-eabi-4_7-2013q2/bin/arm-none-eabi-gcc -fsyntax-only -mcpu=cortex-m7 -mthumb -xc /dev/null >/dev/null 2>&1; echo $?
1
karlp@cuth:~/src/libopencm3 (master)$ ~/tools/gcc-arm-none-eabi-4_9-2015q2/bin/arm-none-eabi-gcc -fsyntax-only -mcpu=cortex-m7 -mthumb -xc /dev/null >/dev/null 2>&1; echo $?
0
karlp@cuth:~/src/libopencm3 (master)$ 
```

If you're toolchain is doing somethign altogether different, then we have a problem.  Particularly, the problem would be returning a return code of 0, but still actually refusing to compile m7 code.

MightyPork, 2015-11-02:
The code that *is supposed to* print the warning "Your toolchain doesn't support -mcpu=cortex-m7" is there, but it does not seem to work.

I'm up to date with origin/master, pulled just today. (Actually, I updated my submodule, which messed up my USB code. While trying to fix it, I run into this bug. Now USB is OK, but this remains...)

MightyPork, 2015-11-02:
@karlp tried that, running

```
arm-none-eabi-gcc -fsyntax-only -mcpu=cortex-m7 -mthumb -xc /dev/null >/dev/null 2>&1; echo $?
```

Gives me return code 0.

karlp, 2015-11-02:
yeah, kuldeep too, that's bogus. also, 5.2 is meant to already support the m7.  He's filed a ticket with arch he says.

kuldeepdhaka, 2015-11-03:
not a libopencm3 issue.
here is the ticket: https://bugs.archlinux.org/task/46951

MightyPork, 2015-11-03:
hm yes. 

but is there some way to make it compile only stm32f3 stuff?
It'll take a while before they fix it, and anyway the lib takes like a minute to build, as it builds all the stuff I don't want.

karlp, 2015-11-03:
well, you only build it once.  If you really must, and really can't wait, you can comment out targets here: https://github.com/libopencm3/libopencm3/blob/master/Makefile#L37-L41

closed 2015-11-03 by karlp

MightyPork, 2015-11-03:
I was thinking maybe if the line used `TARGETS =? ` or something, I'd be able to override it when running the makefile. Not sure.

It's in a git submodule, so editing it isn't practical

karlp, 2015-11-03:
You can make a pull for that feature, I very much like people actually at least _compiling_ everything.  Note that you can have your own submodule anyway, there's no requirement that the submodule be the github.com/libopencm3/libopencm3 repository itself.  If it's only about the speed of a personal build, you can just have a local commit there.

You can also use "make -j 8" or so, and you can, as I suggested, remember that you only need to build it once.

MightyPork, 2015-11-03:
I've sent the pull request: https://github.com/libopencm3/libopencm3/pull/559
By default there's no change, but you can now limit it to specific target, if you feel like it.