libopencm3/libopencm3 #560: STM32F3 rcc error

baoshi, 2015-11-04:
It is possibility a typo in lib/stm32/f3/rcc.c

void rcc_set_ppre2(uint32_t ppre2)
{
	uint32_t reg32;

	reg32 = RCC_CFGR;
	reg32 &= ~((1 << 13) | (1 << 14) | (1 << 15));
	RCC_CFGR = (reg32 | (ppre2 << 11));
}

void rcc_set_ppre1(uint32_t ppre1)
{
	uint32_t reg32;

	reg32 = RCC_CFGR;
	reg32 &= ~((1 << 10) | (1 << 11) | (1 << 12));
	RCC_CFGR = (reg32 | (ppre1 << 8));
}

I believe the bit number of bitwise &= is incorrect. PPRE1 is RCC_CFGR[10:8] and PPRE2 is RCC_CFGR[13:11]

karlp, 2015-11-04:
Yes, this is correct.  It's in my development tree for F3 already: https://github.com/libopencm3/libopencm3/pull/555  I'll try and reshuffle and get this particular commit in earlier, it's more important obviously!

baoshi, 2015-11-04:
Would you also please add the missing:
void rcc_set_prediv(uint32_t prediv)
{
    uint32_t reg32;

    reg32 = RCC_CFGR2;
    reg32 &= ~(RCC_CFGR2 << RCC_CFGR2_PREDIV_SHIFT);
    RCC_CFGR2 = (reg32 | (prediv << RCC_CFGR2_PREDIV_SHIFT));
}

karlp, 2015-11-04:
fixes for both will land in a few days, see #561

karlp, 2015-11-08:
this fix has now landed.

closed 2015-11-08 by karlp