libopencm3/libopencm3 #565: Missing function to clear USART interrupt flags in  F3

MightyPork, 2015-11-13:
There is a function to check interrupt source, to read a flag, but no function to clear a flag.

Clearing those flags is done by writing ones `USART_ICR(usart_base)`.

Currently I'm using `USART_ICR(com_dev) = 0xFFFFFF;` to clear all flags, but I'd expect a function if there's a function for reading said flags.

karlp, 2015-11-13:
Have a look at f0 if someone's added it there yet.  The f3 (and then f0, l0, l4) introduced a new uart peripheral, the "older" uart on f1,f2,f4,l1 doesn't have the ICR register at all, hence the lack of heler function.  f3 is not a heavily loved platform much (yet) so some of these newer features simply haven't had anyone add them yet unfortunately

ChuckM, 2015-11-19:
In general, a function to change the flags would be problematic. A number of subtle bugs have been introduced over the years as the ARM read/modify/write code and 'reset on read' flags have collided. Since this operation is generally called in time critical interrupt service routine code, it makes no sense to add a call/return operation simply to reset a flag. 

Additionally you don't generally want to clear *all* the flags as you may find that later processors have defined flags that you care about later.  You can clear all the flags that are set by reading the flag register and writing it to the clear register. A useful idiom here is 

```
void usartx_isr(void)
{
    uint32_t flags = USART_ISR(usart_base);
    USART_ICR(usart_base) = flags; /* clear all the set flags */
    if (flags & USART_ISR_TXE) { ... process transmit ... }
    if (flags & USART_ISR_RXNE) { ... process receive ...}
    return;
}
```