libopencm3/libopencm3 #587: Wrong VREFINT_CAL & TSENSE_CAL addresses for STM32L100

MightyPork, 2015-12-30:
File: https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/l1/memorymap.h#L117

```
#define ST_VREFINT_CAL			MMIO16(0x1FF80078) //<-- wrong
#define ST_TSENSE_CAL1_30C		MMIO16(0x1FF8007A) //<-- wrong
#define ST_TSENSE_CAL2_110C		MMIO16(0x1FF8007E) //<-- wrong
```

At least on STM32L100RC, the addresses end with `Fx`, not `7x`.
The `7x` addresses hold just zeros.

Dump for reference: https://gist.github.com/MightyPork/64efc7202cd5a3b4d957

I'm not sure if this applies also to other L1 devices. The datasheet doesn't mention the temperature calibration constants (a bug probably), but the voltage reference is there.

karlp, 2015-12-30:
dupe of #234 most likely.  same problem anyway, no separation of l100 and l15x

karlp, 2015-12-31:
L100 Low and Medium match the existing code, it's the same M+ issue as with l15x.. (so...xxxC parts will be "wrong" and xxx8,B parts will be "right")

If you're looking for this in the datasheet again, (I saw you had a hard time finding it) it's in section 6.3.3 of rev4 of the datasheet for both the L,M and M+ devices.

Regardless, there's no temp sensor documented as existing at all :)

MightyPork, 2016-01-01:
yea, the sensor is there but doesn't work very well (more like, utterly broken). So guess it doesn't matter that much

karlp, 2020-03-06:
I'm reopening this, because I've closed #234 by other work, but it doesn't and won't fix this issue in that manner.