libopencm3/libopencm3 #590: STM32F3 missing opamps (attached code that can be included)

MightyPork, 2016-01-05:
I needed opamps in STM32F303, so here's my library files to configure them:

Files are in this repo: https://github.com/MightyPork/libopencm3_stm32f3_opamp

May work for other F3 models, this is the only one I use, so I didn't research others.

I'm not sure how to properly add it to the library, so I'll leave that to someone more competent.

The `clock.h` is just a stub, because the calibration routine needs a delay_ms() routine, which I derived from SysTick. That will have to be done some other way in the library.

karlp, 2016-01-05:
It's substantially easier for people to review these if you just make it as a pull request.  

You just put the files in, say, the lib/stm32/f3 directory and add them to the lib/stm32/f3/makefile.

For things that need delays, the "best" path forward is to make a routine that does something, a routine that checks whether it's ready or not, and then the second part if necessary.  People have all sorts of different methods of waiting, and blocking 2ms waits isn't normally very popular :)

something like "opamp_calibrate_start()" and "opamp_calibrate_done()" maybe?

MightyPork, 2016-01-05:
okay I'll give it a go when I have more time (busy end of semester).

Not sure with the delays, as it's in a for loop. 

It waits for some sort of internal capacitor to charge, so there's no indication that it's done.
However, it's afaik done only once when you configure the opamp.

Maybe a weak prototype in the .h file, that people can implement? They could also choose not to calibrate, and thus leave it undefined.