libopencm3/libopencm3 #594: GD32F103 support

rgwan, 2016-01-07:
I have created a branch for GD32F103 support. The GD32F103 is very similar to STM32F103.

The big difference is the USB scaling bits, PLL/EXOSC start time, and flash programming register. Early version of GD32 has a UART bug, but it is not big problem, it can be omitted.

The GigaDevice not only has 100 series, but also has 200 series(but it doesn't like STM's 200 series), and 150 series(like STM32F051, but it uses M3 core).

I will be glad if upstream can merge my WiP branch~

Zhiyuan

rgwan, 2016-01-07:
https://github.com/rgwan/libopencm3/tree/gd32-wip

Here is my working in progress tree.

dword1511, 2016-01-23:
I am wondering what are the necessary changes. I tried just to use the STM32 version of libopencm3 on GD32 except writing my own clock setup, and it seems to be fine, until I discovered that SPI does not work.

bikemike, 2017-04-26:
@rgwan, what is the status on your gd32 branch?

rgwan, 2017-09-06:
@bikemike Well, I'm prepare for my postgraduate entrance. I will stop my porting for about 1yr. I'm sorry to tell you that.

karlp, 2019-06-03:
gd32f1x0 has landed, which is indeed not the same as the f103, but it might provide an easier starting point for anyone wanting to work on this further.  PRs are welcome here, but I don't see a lot of benefit to keeping this open as a _ticket_ at this point.

closed 2019-06-03 by karlp