libopencm3/libopencm3 #600: [USB-OTG] Control FIFO out word loss

H2OBrain, 2016-01-17:
When receiving control out data in usb-otg (hs and fs) the data is shifted by 4 bytes (1 word) so that the first byte received is the actual fourth byte and the last four bytes are just garbage. 

Example : 
```
PC  CTRL_OUT  size: 11b  msg: 01234567891
MCU CTRL_OUT  size: 11b  msg: 4567891asdf
```

To fix this issue a delay was added in usb_fx07_common.c (see FIXME: Why is a delay needed here?)
This delay causes the whole USB device to crash when I run my adapted (control loopback test)[https://github.com/H2OBrain/libopencm3/commit/e57ca422d92bb7ed49487ffe08f10036c30eda56].
Without this delay, the device 'never' crashes but the first four bytes are always lost.

There's an old thread on st's forum about this [link](https://my.st.com/public/STe2ecommunities/mcu/Lists/cortex_mx_stm32/Flat.aspx?RootFolder=%2Fpublic%2FSTe2ecommunities%2Fmcu%2FLists%2Fcortex_mx_stm32%2FUSB%20FS%20Losing%20word%20in%20RxFIFO%20%28STM42F40x%29&FolderCTID=0x01200200770978C69A1141439FE559EB459D7580009C4E14902C3CDE46A77F0FFD06506F5B&currentviews=1939)

I tried a couple things myself with no success so far.

karlp, 2016-01-17:
sounds like an _excellent_ case for the tests to try handling.

prusnak, 2017-03-19:
Is the issue fixed by fixes in https://github.com/libopencm3/libopencm3/pull/672 ?

H2OBrain, 2017-03-19:
@prusnak Yes it is fixed by #672.

alibabashack, 2017-12-17:
Hi, I'm sorry to say that there is a regression here. I just ran a git bisect against my implementation of a USB DFU loader on a STM32F107 and commit 58f2ee34fa7b6fc04c41acff221d7e916467f7b9 (usb: Ensure control events are handled in order by @zyp ) is reported as the breaking commit. Could someone please take a look?

karlp, 2017-12-17:
@alibabashack could you please make this a separate issue then?  THis PR wasn't merged, so I'm not sure what you're talking about with a regression.

As always, if you have a usb concern, the ultimate best way of getting it looked at is to make a testcase using the gadget0 test framework, or at least provide _some_ mechanism for reproducing your issue.   Your current statements don't contain nearly enough information to do anything with.

zyp, 2017-12-17:
For reference, 58f2ee3 is from PR #785. I agree with @karlp that the best way of handling this would be to create a new issue report for it, referring to #785.

To be able to debug it, we also need a test case that demonstrates the problem. Simply stating that you have a problem that goes away when you revert 58f2ee3 gives us no clue about what can be wrong. For all we know, the problem could be that your code relies on assumptions that used to be true but now are false.

When you report the new issue with a test case, please also include a detailed description of what you have observed. What did you do to make it happen? Does it happen every time? If not, how often?

alibabashack, 2017-12-17:
@karlp, @zyp 
> could you please make this a separate issue then? THis PR wasn't merged, so I'm not sure what you're talking about with a regression.

Sorry, but as I see it this ticket is not a merge-request, but a bug report as the title suggests. Other tickets which fixed the bug were referenced and the bug reporter confirmed that the bug was fixed (at time of writing), which is true as far as I can reproduce. This is when this ticket could have been closed. In this sense I am now reopening the ticket, because I can reproduce the exact defect as initially described, which makes it a regression. Therefore I hope it is ok that I continue here, because a new ticket would be nothing more than a duplicate:

To reproduce please take a look at my branch https://github.com/alibabashack/libopencm3/tree/bug-usb-transfer-word-skipped. I ported the gadget-zero tests to my target board (since no board containing an stm32f107 seems to be available), cherry-picked a loopback test originally written by @H2OBrain and reverted the defective commit 58f2ee3. To be honest, I did not look too much into the test logic, but it reproduces my exact problem of loosing the first 4 bytes of some control transfers. Maybe in doubt @H2OBrain can comment on the test logic? Using this tree I could reproduce both, the bad and the good behavior, as follows:

# prepare
`cd your/libopencm3`
`git remote add alibabashack https://github.com/alibabashack/libopencm3.git`
`git fetch alibabashack`

# bad
`cd your/libopencm3`
`git checkout alibabashack/bug-usb-transfer-word-skipped-bad`
`make clean`
`make all -j24`
`cd tests/gadget-zero`
`make -f Makefile.waveshare-open107v clean all flash V=1`
`pyvenv .env`
`. .env/bin/activate`
`pip install pyusb`
`python test_gadget0.py`
This gives:
```
Running tests for DUT:  waveshare-open107v
.........ss.F...............
======================================================================
FAIL: test_loopback (__main__.TestControlTransfer_Reads)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test_gadget0.py", line 299, in test_loopback
    for i in range(4, ep0_size) : inner(i)
  File "test_gadget0.py", line 296, in inner
    buf_out.tobytes(), buf_in.tobytes()))
AssertionError: array('B', [52, 53, 54, 55, 56, 57, 48, 49, 50, [35 chars], 1]) != array('b', [48, 49, 50, 51, 52, 53, 54, 55, 56, [39 chars] 57]) : Buffers don't match!
 - buf_out : b'01234567890123456789'
 - buf_in  : b'4567890123456789\x00\x00\x07\x01'

----------------------------------------------------------------------
Ran 28 tests in 0.461s

FAILED (failures=1, skipped=2)
```

# good
`cd your/libopencm3`
`git checkout alibabashack/bug-usb-transfer-word-skipped-good`
`make clean`
`make all -j24`
`cd tests/gadget-zero`
`make -f Makefile.waveshare-open107v clean all flash V=1`
`pyvenv .env`
`. .env/bin/activate`
`pip install pyusb`
`python test_gadget0.py`
This gives:
```
Running tests for DUT:  waveshare-open107v
.........ss.................
----------------------------------------------------------------------
Ran 28 tests in 0.541s

OK (skipped=2)
```
I hope this helps to understand the bug. I am willing to run patches on my board, just drop me a message.

Regards, Alex

zyp, 2017-12-18:
Ah, now I follow, I didn't read the entire context earlier.

I still think that a regression is better reported as a new issue (with a reference to the previous). While it might be the same problem, the code causing it now is different from the code causing it before and would require a different fix.

For reference, as far as I can recall at the moment, this issue were caused by a race condition in the usb hardware. It was fixed in 0c1ff68 which is rebased from #672, by rearranging the order of some operations related to flow control (CNAK/SNAK).

#785 enables the use of flow control on EP0 to ensure that control events are handled in order. It is probably adding flow control operations at a point that hits the race condition again.

If I have time tomorrow, I'll look into the involved code and see what I can find.

zyp, 2018-01-19:
I've opened #873 to track the regression. Please take further discussion there.

closed 2018-01-19 by karlp