libopencm3/libopencm3 #602: add doxygen stm32 common functions output for stm32f1

tsaitgaist, 2016-01-25:
The common functions (STM32) are missing in the documentation for the STM32F1.
I've attached a patch to fix this issue in Doxygen files.
I did not check if this applies to other families as well.

[0001-add-doxygen-common-output-for-stm32f1.txt](https://github.com/libopencm3/libopencm3/files/103156/0001-add-doxygen-common-output-for-stm32f1.txt)

ksarkies, 2016-01-25:
The doxygen structure is seriously complicated by having to be forced to 
generate multiple documents, something that it wasn't designed to do. 
Maybe doxygen wasn't really the best choice. Anyway the structures put 
in place turned out to be rather delicately balanced. Please check that 
this change doesn't create major issues, in particular functions 
occurring multiple times or in unexpected and unwanted places. It may 
well be that some functions are missing, but that could indicate a 
subtler problem or simply new API additions or changes to loc3 that 
haven't been properly added into the documentation.

tsaitgaist, 2016-01-26:
I'm not sure I can check this and how. I'm not familiar with the libopencm3 documentation structure, but more importantly it was my first go at doxygen.
I was wondering why the STM32F1XX Timer section in the documentation only included the function defined in lib/stm32/f1/timer.c but not lib/stm32/common/timer_common_all.c. By trial and error I solved this issue, but I don't know all implications. That's also why I opened an issue instead of submitting a patch.

ksarkies, 2016-01-27:
> I'm not sure I can check this and how. I'm not familiar with the
> libopencm3 documentation structure, but more importantly it was my
> first go at doxygen.
> I was wondering why the STM32F1XX Timer [...]
Given its complexity it would be best to just signal problems like the 
one you mention above. I had in mind to check over the doxygen again 
sometime soon and do some repairs. This is a good reminder to do it. 
Thanks for raising this.

tsaitgaist, 2016-02-16:
to be more compliant to github I created pull requestion #623

ildar, 2017-03-09:
Looking at the project layout I see that it is file-level granular. I mean that e.g. STM32F1 library is assembled of files that are located in a few directories, see https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/f1/Makefile#L42 . The disorder here is in the fact that some files from the .../f1/ directory are used, some are not, AND some files from some other directories are used too, but not all (look at VPATH)
With the current state of source files layout, Doxygen should be set to process particular files from the couple of directories.
Hence, @tsaitgaist , the flaw of your patch is that it is directory-level and doesn't comply to the real file list that's built. But IMHO it makes things better allowing better docs then they are now.

karlp, 2017-10-18:
This should all be fixed now, I just landed a patch that generates the doxygen source files based on the makefiles and the .d files instead of manually maintaining them

closed 2017-10-18 by karlp