libopencm3/libopencm3 #621: flash size info for STM32F1 is 16 bits instead of 32

tsaitgaist, 2016-02-16:
According to STM32F1 reference manual RM0008 section 30.1.1 the flash size register is 16 bits.
/lib/stm32/f1/flash.c reads a 32 bits value:
#define MEMORY_SIZE_REG            MMIO32(DESIG_FLASH_SIZE_BASE)
instead a 16 bits values should be read:
#define MEMORY_SIZE_REG            MMIO16(DESIG_FLASH_SIZE_BASE)
this 32 bits value also induces some error in the code since more is done when MEMORY_SIZE_REG is bigger than 512, which it is when a 32 bit value is read (not sure what the first 16 bits stand for though).

In the attached patch I also took the liberty to move this definition to the header since I can be useful for the user for range checks when using the library.

[0001-fix-flash-size-is-a-16-bits-values-move-flash-size-i.txt](https://github.com/libopencm3/libopencm3/files/132177/0001-fix-flash-size-is-a-16-bits-values-move-flash-size-i.txt)

karlp, 2016-02-16:
can you elaborate on what th bad behaviour is please?  I agree it should be 16bit.

Can you please make a regular github pull reqquest? seeing as you're already a github user?  Can you please make patches with  more standard git commit messages? http://chris.beams.io/posts/git-commit/

tsaitgaist, 2016-02-16:
is #622 fine?

karlp, 2016-02-16:
The correct fix here was to use the existing DESIG_FLASH_SIZE macro, available for all stm32s, rather than this duplicate and broken define. I've pushed a fix for it.

closed 2016-02-16 by karlp