libopencm3/libopencm3 #626: STM32 USB PMA memory overlap

dmitrystu, 2016-02-23:
in function  st_usbfs_set_ep_rx_bufsize found incorrect calculation for NUM_BLOCKS
In case size < 62 NUM_BLOCKS should be size/2 because block is 2-byte wide. Therefore  actual memory reserved by usb machine is double that considered in pm_top variable. (For size > 32 also BLSIZE set)
Also in case size > 62  actual reserved memory is large then considered in pm_top.
This will cause a PMA memory overlaping.
[stm32_pma_overlap.diff.txt](https://github.com/libopencm3/libopencm3/files/143023/stm32_pma_overlap.diff.txt)

karlp, 2016-02-23:
See https://github.com/libopencm3/libopencm3/pull/477

karlp, 2018-08-20:
This was fixed via #880 and friends, which has (finally) merged along with tests

closed 2018-08-20 by karlp