libopencm3/libopencm3 #644: are you sure about the GPL style license?

hackrid, 2016-04-07:
Hi,

the gpl license of this projects makes it impossible to use and therefore improve it.
did i miss a static link exception which is mandatory for bare-metal systems?

karlp, 2016-04-07:
no, you didn't.  it's LGPL, you need to provide a way for the binary to be relinked.  This can be providing your makefile and binary .o files, and allowing the user to relink their own build of libopencm3.  If, in consultation with your legal advice, this is unsuitable for you, I'm sorry.  This project has reached a state where relicensing, while very desirable to some community members, is very undesirable to others.    Please note further, that these days, the ARM CMSIS header files are actually distributed under proper permissive licenses, so the direct _need_ for this project to even exist are less clear than they were 6-7 years ago, and you may be perfectly well suited with using CMSIS directly.

closed 2016-04-07 by karlp

hackrid, 2016-04-08:
Hi Karl,

thanks for your write-up.

I was looking for a framework to write portable software across mcu vendors. CMSIS also does not offer this level of abstraction. do you have something in mind which could be used?

Best regards

karlp, 2016-04-08:
perhaps mbed.