libopencm3/libopencm3 #649: stm32-f7 build: gcc check is insufficent

quitte, 2016-04-09:
The Makefile for stm32-f7 checks wether the installed gcc supports cortex-m7. However it does not check wether the installed binutils support cortex-m7.
This is an issue on ubuntu systems, where arm-none-eabi-as version 2.25 is paired with arm-none-eabi-gcc 4.9.3, which causes the build to fail despite the check.
Installing binutils-arm-none-eabi 2.26 of ubuntu xenial fixes the build.

karlp, 2016-05-06:
We spoke about this on irc, just noting it here for reference.  I don't really consider this to be a case that I feel we need to consider.  I strongly feel this is a critical issue with ubuntu packaging, not something we should have to try and detect.  If the "arm-none-eabi-gcc" in the path supports m7, but the -ld and -as don't, I _strongly_ feel that to be a user problem, not something the project is meant to  somehow detect specially.  I'm inclined to close this and the pull checking for it, but am open to other opinions.

karlp, 2017-04-24:
As above, and as per comments on PR #650, I think we should close this.

closed 2017-04-24 by karlp