libopencm3/libopencm3 #654: stm32f0: ADC calibration never finishes when adc_power_off is called before

DanielG, 2016-04-16:
The following code will hang forever in `adc_calibrate_wait_finish`:

```
rcc_periph_clock_enable(RCC_ADC);
adc_power_off(ADC);
adc_calibrate_start(ADC);
adc_calibrate_wait_finish(ADC);
```

whereas this works just fine:

```
rcc_periph_clock_enable(RCC_ADC);
adc_calibrate_start(ADC);
adc_calibrate_wait_finish(ADC);
```

RM0091 says to "ensure ADCEN=0" which is implemented in the example code from the appendix as follows:

```
/* (1) Ensure that ADEN = 0 */
/* (2) Clear ADEN */
/* (3) Launch the calibration by setting ADCAL */
/* (4) Wait until ADCAL=0 */
if ((ADC1->CR & ADC_CR_ADEN) != 0) /* (1) */
{
    ADC1->CR &= (uint32_t)(~ADC_CR_ADEN); /* (2) */
}
ADC1->CR |= ADC_CR_ADCAL; /* (3) */
while ((ADC1->CR & ADC_CR_ADCAL) != 0) /* (4) */
{
    /* For robust implementation, add here time-out management */
}
```

Looking at `adc_power_off_async` it seems to me that maybe it should check if ADEN=1 before trying to power down the ADC by setting ADDIS or maybe `adc_calibrate_start` should just force ADEN=0 like the example code does?

For easy reference:
```
void adc_power_off_async(uint32_t adc)
{
	uint32_t checks = ADC_CR_ADSTART;
	uint32_t stops = ADC_CR_ADSTP;
#if defined (ADC_CR_JADSTART)
	checks |= ADC_CR_JADSTART;
	stops |= ADC_CR_JADSTP;
#endif
	if (ADC_CR(adc) & checks) {
		ADC_CR(adc) |= stops;
		while (ADC_CR(adc) & checks);
	}
	ADC_CR(adc) |= ADC_CR_ADDIS;
}
```

(I'm testing this on a STM32F042C6T6)

I'd be happy to submit a patch if someone can tell me what the preferred fix is.

kungjohan, 2016-04-20:
2016-04-16 17:06 GMT+02:00 Daniel Gröber <notifications@github.com>:

> The following code will hang forever in adc_calibrate_wait_finish:
>
> rcc_periph_clock_enable(RCC_ADC);
> adc_power_off(ADC);
> adc_calibrate_start(ADC);
> adc_calibrate_wait_finish(ADC);
>
>
>
I have noticed the same behaviour on my STM32F030.
Previous application code worked, but now it hangs with latest libopencm3
(master for current date) .
However it is working with the commit ca50f069b68a39 of libopencm3.
So for now it seems like head is broken.

Best regards
Johan

karlp, 2016-04-20:
@kungjohan can you give me some sample application code you're using that is being affected by this? Just curious what you were doing.  WIll definitely look at this as a bug to fix, no dispute

kungjohan, 2016-04-21:
I can reproduce this error with 5b35c7d7a1be8fefd9caf072a4542f9bada0d33f
libopencm3-examples/examples/stm32/f0/stm32f0-discovery/adc/adc.elf

Rev
d3fff11c1f68b706591c0d51c82d18a0bc88dc17
of libopencm3

I download this to my STM32F0308-DISCO board and run with gdb.
(gdb) bt
#0  0x080003ee in adc_calibrate_wait_finish (adc=adc@entry=1073816576) at
adc.c:418
#1  0x08000106 in adc_setup () at adc.c:42
#2  main () at adc.c:115

Best regards
Johan

elemhsb, 2016-05-10:
Double power off hangs stm32f303vct at stm32f3-discovery while poweron adc.
My solution here is:
```
diff --git a/lib/stm32/common/adc_common_v2.c b/lib/stm32/common/adc_common_v2.c
index ae4008a..a6a7cf4 100644
--- a/lib/stm32/common/adc_common_v2.c
+++ b/lib/stm32/common/adc_common_v2.c
@@ -138,8 +138,8 @@ bool adc_is_power_off(uint32_t adc)
  */
 void adc_power_off(uint32_t adc)
 {
-       adc_power_off_async(adc);
-       while (!adc_is_power_off(adc));
+       while (!adc_is_power_off(adc))
+               adc_power_off_async(adc);
 }
 
 /**
```
it solves also issue libopencm3/libopencm3-examples#133

ha-ko, 2016-05-29:
Just found that there was already an open issue for that. I stumbled upon the same problem and fixed it in my version just as DanielG proposed (in congruence with the datasheet) by checking if ADEN=1: https://github.com/libopencm3/libopencm3/pull/662
That works fine and is imho the correct approach. I'll leave the pull request open for reference for now.

karlp, 2016-09-09:
Fixed via merge of c2e0afdb057ccf4c6440322a1409ec7b94fe16ea
tests added in https://github.com/karlp/libopencm3-tests/commit/55cf5bcd76af3d8137ee0d34ff4507c122540daa  (I also locally tested with the calibrate_wait() code you had.

Sorry this was broken for so long.

closed 2016-09-09 by karlp