libopencm3/libopencm3 #655: stm32 timer_set_period() is off by one

tormodvolden, 2016-04-17:
This is a characteristic of the STM32 timers: The TIMx_ARR register is the last value counted to before going to zero again, so the period is TIMx_ARR + 1.

Therefore the timer_set_period() should either subtract 1 from the passed period value in timer_common_all.c, or the documentation should explain that the actual period will be the value + 1 (and lot of examples should be updated too).

I would be happy to provide patches for either, if I know which variant suits the project best.

Note that the same +1 characteristic applies to the prescaler value, but here it is documented in the code,
"The timer clock is prescaled by the 16 bit scale value plus 1", although many examples are wrong or not clear about this.

karlp, 2016-04-17:
isn't this just a case of "set_period" sets an ARR reload value, not any explict Hz rate, so you put in the value you the user want to use, knowing the reference manual?

tormodvolden, 2016-04-17:
Yes, but it is hiding this from the user by calling it "period". The reference manual never says the ARR reload value will be the period. It depends how general you would like to make these functions - should it be compatible with a possible timer_set_period() on another platform where "period" means what you expect it to be?

If you keep it meaning the ARR reload value, I think at least it should have a comment like the one for the prescaler, and that many examples need fixing up because they assume "period" is the resulting period. Example: timer_set_period(TIM1, 72000000 / 32000);

plaes, 2016-04-21:
Yeah.. confirmed at least on stm32f103 TIM2.

karlp, 2016-04-22:
I'd like to see a broader discussion on this whether the "-1" should _always_ be included inside the library code or not.  I very much don't want a situation where some calls have -1 and some don't.

ChuckM, 2016-07-28:
My take on it is that the field in the register takes the period -1 so you should pass in the period after you have already subtracted 1. However, in the helper function it makes sense to just use the period value and do the subtraction in the helper. Given the register define nature of the library the first solution seems to be the choice with the least surprise of existing users.

karlp, 2016-11-21:
@ChuckM which one is "first" ?

A) all timer_xxx functions that go to a register with a -1 in it (prescaler, reload variables, you name it) take a parameter _without_ subtracting (ie, set_period(TIM1, 72e6/32e3)
B) all timer_xxx functions that go to a register with a -1 in it (prescaler, reload variables, you name it) take a parameter _with_ subtracting (ie, set_period(TIM1, (72e6/32e3)-1)

B is clearly "nicer" from the user point of view, but should all parts of the code be made to do this? or just "set_period" functions? (that seems confusing)

ChuckM, 2016-11-21:
@karlp I've come down on the 'option B' side in my code. My reasoning was as follows;

Not all timers, or divisors, internally subtract 1 from the value in the field, STM32 is inconsistent (which is probably dependent on where they got the peripheral design from at the time). I made too many mistakes where I didn't remember that "this" divisor was a -1 divisor, since most of them seem to not be that way. As a result I decided at my code level *all* divisors passed in would be as expected by the programmer (and consistent) and if the peripheral was treating it as one less than the actual divisor I would do the subtraction inside the library function. It gives me one less think I have to remember and that allows me to screw up less writing code.

karlp, 2016-11-30:
So, Prescaler is always -1, I can see the case for the library to automatically do the substraction, so the _user_ code looks like this...
```
//old prescaler for 1usec ticks
timer_set_prescaler(TIM2, (rcc_apb1_frequency / 1e6)-1);
//new prescaler for 1usec ticks
timer_set_prescaler(TIM2, (rcc_apb1_frequency / 1e6)); 
```

But what about timer_set_period?  If you do the -1 in the library, how do you set it to 0xffff full range? set_period(0)?  That doesn't seem correct anymore.

FabianInostroza, 2016-12-11:
0x10000?, if the parameter is (u)int32 that would give the correct result and interpretation