libopencm3/libopencm3 #656: make STM32F1 lib fatal error: libopencm3/stm32/f1/nvic.h: No such file or directory

lzq888, 2016-04-29:
hi 
    I try to make stm32f1 lib
    it dispaly message  fatal error   : 
////////////////////////////////////////////////////////////////////////////////////////////////////////////////    
@ubuntu:/home/e/libopencm3/lib/stm32/f1$ make all
  CP      libopencm3_stm32f1.ld
  CC      adc.c
  CC      adc_common_v1.c
  CC      can.c
  CC      desig.c
  CC      ethernet.c
  CC      flash.c
  CC      gpio.c
  CC      rcc.c
  CC      rtc.c
  CC      timer.c
  CC      mac.c
  CC      mac_stm32fxx7.c
In file included from ../../../include/libopencm3/cm3/nvic.h:133:0,
                 from ../../ethernet/mac_stm32fxx7.c:38:
../../../include/libopencm3/dispatch/nvic.h:8:39: fatal error: libopencm3/stm32/f1/nvic.h: No such file or directory
compilation terminated.
make: *** [mac_stm32fxx7.o] Error 1

tidklaas, 2016-04-29:
You are not supposed to start the build process from inside a target's directory. If you want to only build a specific target's library, call make from locm3's top level directory and use the TARGETS variable:
`make TARGETS=stm32/f1 all`

karlp, 2016-04-29:
yep, this isn't a supported thing.  closing this.

closed 2016-04-29 by karlp