libopencm3/libopencm3 #659: Where is stm32L4 support ?

Salamandar, 2016-05-17:
Well I can see the L4 support is on its way (seeing the last commits). But where is it exactly ?
The `libopencm3_stm32f4.a` file is not linked with `usart_common_all.o`, is it normal ?
My stm32f4 project does not compile because of the USART function missing… And the rcc ones : `rcc_clock_setup_hse_3v3`and  `rcc_hse_8mhz_3v3`… And that is a big issue ! ^^

karlp, 2016-05-17:
L4 is still in early days, needs more users, more committers, more pulls, more time.  Your other question seems to be about f4?  for any target, higher and higher level functions arrive as they are tested and integrated, there's no full time developers working on this for you.

closed 2016-05-17 by karlp

Salamandar, 2016-05-17:
I meant `libopencm3_stm32l4.a`, sorry :p
Well, how much is the L4 code different from the F4 or the L0 ?

karlp, 2016-05-17:
some similar, needs testing and integration, some very different.  You can find all this in the reference manuals yourself.

Salamandar, 2016-05-17:
Okay, i'll check it out !