libopencm3/libopencm3 #660: STM32F303x6/x8 usart_set_baudrate uses incorrect clock

jschall, 2016-05-19:
See clock tree: http://www.st.com/content/ccc/resource/technical/document/datasheet/group3/38/c6/ca/d1/f7/d5/4e/30/DM00092070/files/DM00092070.pdf/jcr:content/translations/en.DM00092070.pdf

usart_set_baudrate tries to use rcc_apb2_frequency instead of rcc_apb1_frequency with -DSTM32F3

karlp, 2016-09-13:
Valid, one of many braindead things in the f3 series from ST's end.  Only applies to USART1 though, usart2,3 are correctly on apb1.

ChuckM, 2016-09-13:
In general the set baud rate function is a bit limited. Given that a number of USART/UARTs can choose different clock sources for their baud clock (this is especially important in the LP USARTS which use the LSE for their baud clock) that it might make sense to have a function in the USART code that is `usart_get_clock(USARTx)` which would return the frequency of the selected clock, as configured in the register setup.

karlp, 2018-05-02:
I think this happens with some of the f4 usarts too...

stoyan-shopov, 2020-11-05:
I think there is a similar case, at least for STM32F72x and STM32F73x devices, where the U(S)ART have the exception, that their clocks can be selected from several sources. In that case `usart_set_baudrate ()` will not work.

I could prepare some pull request, but I am not sure what will be acceptable as a solution to detect such exceptions.

karlp, 2020-11-06:
The pac55xx PR and some related f7 stuff pulls in a bunch of stuff to lookup the current peripheral clock and return that, it's kinda invasive, but it's likely to be the only way forwards that's sane.

tomldavis, 2022-01-28:
Probably same issue for the STM32F302. With 72MHz clock via external xtal, I get correct the baud rate with:
`usart_set_baudrate(USART1,2*baud);`.