libopencm3/libopencm3 #661: Use of inline functions

ha-ko, 2016-05-24:
Hi,

is there a particular reason why none of the short functions in libopencm3 are defined as inline functions?
E.g. 
void timer_clear_flag(uint32_t timer_peripheral, uint32_t flag)
{
	/* All defined bits are rc_w0 */
	TIM_SR(timer_peripheral) = ~flag;
}
in /stm32/common/timer_common_all.c .
As it's mostly used in interrupt routines these calls should be as fast as possible imho.
gpio_set etc. the same.

karlp, 2016-05-24:
mostly because everyone has different ideas of what should or shouldn't be inlined, and the compiler can just figure it all out if you go to LTO.  it also has some non-zero impact on the ability of a user to relink their end binary against revised libopencm3 so it has lgpl compliance impacts.  People occasionally complain about this, but the vast majority of people don't actually get any performance impact from it anyway, and the few that do, can do their own thing.

closed 2016-05-24 by karlp

ha-ko, 2016-05-25:
Ok, thanks for the detailed answer. The licence argument makes sense. The performance impact isn't as negligible imho but you're right, if one is concerned one "can do their own thing". :)