libopencm3/libopencm3 #668: usb stack doesn't recover from timeouts.

jhoenicke, 2016-07-04:
Our device sometimes does long computations, in which it currently doesn't call usbd_poll() regularly.  If at this time the usb hid devices are enumerated the OS sends `GET_DESCRIPTOR STRING` requests which timeouts since the usb handler is not called.   Unfortunately afterwards the device will never recover.  Even when usbd_poll is then called regularly, the `GET_DESCRIPTOR STRING` requests are not answered properly.  They also cause long delays in the USB communication.  This causes the device to become almost unusable.

I tried to make some small test that exhibits the problem (you need to adapt it to your hardware though). You need some program on the PC side that causes the STRINGs to be requested while  the device is busy ("usb is off").  These will timeout but also the once that are sent when the device is not busy anymore ("usb is on") will also timeout.

https://gist.github.com/jhoenicke/13dd9d32ee03488993805f122bb840fd

The original bug report is here: trezor/trezor-mcu#98

zyp, 2016-07-04:
The libopencm3 USB stack itself shouldn't have any requirements on how often usb_poll() needs to be called, but it obviously can't do any work while the device is busy with other stuff.

Most USB control requests have a timeout of five seconds (the exceptions to this have even stricter timeouts). When a requests times out, the host controller gives up waiting for the data and leaves the host software to deal with the situation. A device that let standard control requests time out may be violating the USB specification, and the host stack may judge that such a device is defective and give up trying to talk to it.

Are your long computations part of the startup procedure? If so, you could leave the device in soft detach mode while busy, so the host won't start talking to it until it's ready to answer.

jhoenicke, 2016-07-04:
The long computation part is when the user entered his password, so it is in the middle of the process.  I suspect that the device isn't properly replying even when we later poll regularly, since according to usbmon the PC sends the requests but they don't get answered.  

I can also try to analyse the raw wire protocol with a logic analyzer...

jhoenicke, 2016-07-04:
Okay this happens when device is busy:

```
PC: GET DESCRIPTOR STRING 0 --- timeouts after 1 second.
PC: GET DESCRIPTOR STRING 0 --- timeouts after 1 second.
PC: GET DESCRIPTOR STRING 3
Device (which is no longer busy):  Answers with STRING 0!
PC: GET DESCRIPTOR STRING 0 --- timeouts after 1 second (even though usbd_poll is called)
PC: GET DESCRIPTOR STRING 0 --- timeouts after 1 second
PC: GET DESCRIPTOR STRING 1
Device: STALL
PC: GET DESCRIPTOR STRING 0 --- timeouts after 1 second
PC: GET DESCRIPTOR STRING 0 --- timeouts after 1 second
```

My guess is that the stm32f2 queues the SETUP frames for all three requests, and then libopencm3's usb driver is confused because it reads SETUP SETUP SETUP instead of the usual SETUP OUT SETUP OUT SETUP OUT...

the `timeout, timeout, stall` pattern seems to continue for the later requests.

jhoenicke, 2016-07-05:
I'm shooting a bit in the dark, but it turns out that if the device is in the stalled state, it can be recovered by writing 0 to the DIEPCTL0 register (its content at that time is 0x00080000, i.e. pktcnt=1, xfrsiz=0).

I'm still not completely understanding all these USB registers.

zyp, 2016-07-05:
That sounds like a shot in the dark, yes. According to the RM you're not supposed to touch that register when EP0 IN is enabled, and when it's not enabled, touching it shouldn't have any meaningful effect.

Most likely there's a bug in how libopencm3 handles and recovers from aborted control transfers. My guess is an invalid `usbd_dev->control_state` transition, and maybe some other state that needs to be cleaned up.

Either way, blocking the USB stack and causing aborted control transfers in the first place is poor design on your side even if you could recover successfully. You should consider either splitting your computation code into blocks that you could interleave with other work like USB polling, or adding multithreading that can preempt your computation code.

jhoenicke, 2016-07-05:
I'm trying to solve it from different angles.  Yes, you're right, we should call usbd_poll on long running operations, but I feel a bit reluctant to put a usbd_poll into the inner loop of every crypto function.  It's not clear how often we need to call it and if I understand it, if we call it only every say 100 ms, we should then call usbd_poll several times until all interrupts are handled (in case there were several pending).  Building another thread would probably change a lot in the existing architecture and I'm hesitant to do this just for fixing this bug as it may introduce a lot of new ones.  We will certainly put a usbd_poll at least into the long running pbkdf routine.

But I think it is also a bug in libopencm3 that sending two or three SETUP packets in a row before the first is handled, will cause the device to permanently block control packets.  If I understand it right, the code is actually trying to send the reply but ep_write_packet returns without doing anything because DIEPTSIZ0 has non-zero PKTCNT.  This explains why the device timeouts then.  The stall on request string 3 is caused because the language id is not set (since the request string 0 failed).  So the question is: why is the packet count not zero and the packet not sent anyway?  I think the answer is that the stm32 sets the NAK bit internally when it receives a new SETUP packet.

I think the following happened:
```
PC sent SETUP for GET STRING 0 -- timeout
PC sent SETUP for GET STRING 0 -- timeout
PC sent SETUP for GET STRING 2 with invalid language
Device wakes up.
First SETUP packet is received, controller is programmed to send STRING 0.
String 0 is sent by the hardware (as reply to the last SETUP packet).
Second SETUP packet is received by software, controller is programmed to send STRING 0
Third SETUP packet is received by software,  sets the STALL bit since language is wrong.
```

Now there is a packet enqueued, but there is nobody reading it.  When the next SETUP packet comes, the stm32 sets the NAK bit internally, and the software will never clear the NAK bit since it thinks there is still a packet in the queue. 

A fix would be to clear the packet counter in DIEPTCNT when receiving a SETUP packet, since the hardware is obviously not expecting a reply to the previous SETUP packet.  Then write_packet works again and sends the reply.

jhoenicke, 2016-07-05:
But I'm not sure if writing 0 to DIEPTSIZ will also clear the TxFifo.  The stm32 usb documentation is so hard to read.

jhoenicke, 2016-07-05:
Do you think the commit jhoenicke@d610796 is a good fix?  Does one need to be more careful with clearing the registers.  After SETUP was received the core should already have made sure that NAK is set and that the endpoint is disabled.

I tested it and it works for me.  It may still send the wrong SETUP reply when multiple SETUPs are received but at least it recovers later.

zyp, 2016-07-05:
Ah, I think I see the problem. `stm32fx07_ep_write_packet()` looks at DIEPTSIZ.PKTCNT to check if the endpoint is already busy. I think a better check would be looking at DIEPCTL.EPENA, since DIEPTSIZ means nothing when EPENA is cleared.

jhoenicke, 2016-07-05:
But you also need to flush the txfifo if pktcnt is not zero.  I tried it
and if I don't flush the fifo, the replies don't match the requests.

jhoenicke, 2016-07-05:
But it would also be good to not answer outdated requests. I'm don't have
an idea how to easily check this without looking into the receive queue,
though.

zyp, 2016-07-05:
After reading out a SETUP packet, you could check if there's more contents in the RXFIFO and if so read GRXSTSR to check what it is. If it's another SETUP packet, you can just skip further processing of the previous one.

jhoenicke, 2016-07-07:
There is another related but not identical issue.   If I call usbd_polll more often, e.g. every 100 ms, I get a stall and the same hanging packet.

I think the stall is caused by usb_control_out, because it is called *before* usb_control_in after the last packet.   And this is because the logic first checks the RXFLVL interrupt (where the empty out packet has already been received) before it checks the  DIEPINT(0) interrupt.  

If my analysis is correct, it only works if usbd_poll is called after the last IN packet was sent before OUT was received, which requires to hit a 10 µs window. (a control read message has the form SETUP DATA0 <8bytes> ACK IN  DATA1 <value> ACK OUT DATA1 <> ACK).  The window is actually about 70 µs because if I call it before the next SETUP is received, the STALL has no effect.   Still hard to hit.

zyp, 2016-07-07:
I agree with your analysis.

Device is sitting with control_state.state = LAST_DATA_IN, TX completion should call _usbd_control_in() which changes state to STATUS_OUT, then RX completion calls _usbd_control_out() to handle that and terminate the control transaction. Instead _usbd_control_out() gets called first and asserts a stall because LAST_DATA_IN is not a handled state.

The best way to fix this would probably be to keep EP0 OUT in NAK until the data stage is completed so that the status stage package can't get into the RXFIFO until we're actually in the STATUS_OUT state.

jhoenicke, 2016-07-07:
Wouldn't this affect performance by blocking all packets?  In principle we don't care when the in endpoint is free after we sent the last packet of the reply.  We could just go to STATUS_OUT immediately and ignore the DIEPINT interrupt.

zyp, 2016-07-07:
No, it won't affect other endpoints. I considered your approach as well, but it'll be messier and it just masks the root cause instead of solving it.

karlp, 2016-07-07:
likely related: https://github.com/libopencm3/libopencm3/pull/446 and at least part of https://github.com/libopencm3/libopencm3/pull/628

jhoenicke, 2016-07-07:
Yes, https://github.com/libopencm3/libopencm3/pull/446/files fixes the same problem for a different chipset.  I tried if it is enough to put DIEPINT handling before RXVLFL handling and it works when I poll only every 380 ms.

Also there is this ugly FIXME in the code and I wonder if it is related.  Is there some information why it was added?  The patch libopencm3/libopencm3@e0fe433 that introduced this was from @gsmcmullin 

Another question: is there preemptive multithreading support?  If I have to regularly call yield() in my crypto code, it isn't very much different from calling usbd_poll() directly.  Also has anyone ever tried to use real interrupt handler (OTG_FS vector) for usb?  In that case I would probably have to change our usb callbacks to only enqueue messages after minimal processing.

runn1ng, 2016-11-03:
Hello Jochen! It seems you fixed this issue (according to the commit). Is that pushed into upstream?

karlp, 2017-03-19:
To make the tests work for all platforms, what _explicit_ time window should be allowed per spec?  The current magic numbers seem fairly arbitrary?

jhoenicke, 2017-03-19:
Out of my head, the long timeout should be a bit larger than one second, since this is the time after which the control request timeouts.

The short timeout can be few milliseconds.  It is only important that the control request hits the device when the delay is still running.  So it is really the question, how fast does the PC side send the next request. I think everything between 10 and 500 milliseconds should be fine.

closed 2017-03-30 by karlp