libopencm3/libopencm3 #675: Support for STM32F446 and STM32F469

davids5, 2016-07-26:
There are a few specific changes to support the STM32F446 and STM32F469

1) The RCC start up code needs to not clear the PLLR bits. Bits 30:28 PLLR[2:0]: Main PLL division factor for DSI clock
2) In the USB device init OTG_FS_GCCFG ST changed the definition
F4:
[image]

F446
[image]

Note bit 21 is now an 1 is enabled and use to be 1: VBUS sensing not available by hardware

What is the correct way to specialize a version of a CPU in libopencm3?

ChuckM, 2016-07-26:
It is a challenge. If you wanted to write software that worked on both your only choice would be to decide what to do based on CPUID (and not the standard one, the proprietary one) so code like this might work:

```
#define ST_CPUID             MMIO32(0xe0042000);

void
set_vbus() {
    if (ST_CPUID & 0xfff) == 0x434) ||  (ST_CPUID & 0xfff) == <446 id>) {
          OTG_GCCFG |= 1 << 21;
   }
}
```

There is no "variant specific" indicator in the build or the makefiles which you can use and the same library is going to be created for all F4 chips. Note that the ARM architecture manual's SCB for both processor gives the same part number 0xC24. So you can't use that to distinguish. The best of a bad situation is an API call in the USB stack which sets this bit and looks in the CPU's panties to see which kind it is before operating.
}

nschodet-sevenhugs, 2016-07-26:
For USB, what about OTG_CID register? It is 0x2000 on f479/f469.

karlp, 2017-10-26:
The USB issues were committed back in cf80e2bd5e19eadd8fe0e8719070fc59070a3459
The PLLR bits were fixed back in 57c2b00a69f97205313e1c7ab8116ee1893b231e  Sorry I didnt' close this ticket way back then.

closed 2017-10-26 by karlp