libopencm3/libopencm3 #679: USB endpoint double-buffering (is not yet supported)

urjaman, 2016-07-31:
I might end up implementing that (atleast for stm32f0 or whatever v2 usb) since I have a project that could really use it, but the question here is, how would you like it exposed in the API?

A new parameter to usbd_ep_setup sounds like it would break lots of code, but I dont have other great ideas either. A bit in the type stuff would be breaking the isolation between usb stuff and controller stuff (those bits are defined by usb standard so...), etc.

One of my prototypes had an extra function that mangled the EP to double buffered mode (called right after setup), but that seems a bit hardware specific (eg. can you do that easily?) and not very pretty.

kuldeepdhaka, 2016-08-09:
yes, double buffering isnt supported yet.
best (atm) is to maintain a modified version for your usecase.

WGH-, 2017-08-21:
Judging by the documentation, isochronous endpoints on usbfs_v1 hardware unconditionally use double buffering, and thus are currently unusable on libopencm3. See RM0008, "23.4.4 Isochronous transfers".