libopencm3/libopencm3 #701: Using LTO causes C++ constructors of static variables to be lost

WGH-, 2016-09-20:
I don't know for sure whether it's libopencm3 or gcc fault, though.

I've encountered this on gcc 4.9.3 and gcc 5.4.0.

karlp, 2016-09-21:
there's no lto enabled in the currently committed tree of libopencm3, and your entire application needs to be compiled with lto, not just your portion, so this cant' possibly be a problem in libopencm3 itself at this stage.  It's certainly something to look at it, but it's simply not enabled at present.

closed 2016-09-21 by karlp

WGH-, 2016-09-21:
I did enable LTO for libopencm3 as well.

WGH-, 2016-09-21:
It was my mistake, the constructor was not lost. LTO simply caused them to be run in different order, which triggered a bug in my code. It's neither libopencm3 nor compiler bug.

libopencm3 actually works fine with LTO. I only had to patch `ARFLAGS` everywhere.

```
ARFLAGS                = rcs --plugin=/usr/libexec/gcc/armv7m-hardfloat-eabi/5.4.0/liblto_plugin.so
```